betaflight/betaflight #6351: OSD Zenith and Nadir (up and down direction) symbols

Lyonnard, 2018-07-13:
While learning I spent a lot of time understanding when to cut power in power loops and similar aerobatics.

Would it be possible to have two additional symbols for zenith and nadir (the up and down direction) displayed when artificial horizon is active in the OSD configuration?

This would help to evaluate the craft orientation during flips when the artificial horizon and other visual references are outside the screen.

leocb, 2018-07-13:
Would something like this be good?
https://www.youtube.com/watch?v=yr96a10hkXw
I asked him to open a PR here, if he doesn't want to, I'll try to get his permission and open a PR myself (the code is a little down in the video comments), I personally believe the mod he made to the art. horizon is awesome!

Lyonnard, 2018-07-13:
That would definitely be an option. I am not sure that the moving arrow indicating the closest direction to the horizon would be as effective as a point fixed point in orientation as the virtual horizon is, it may be confusing to have it zipping around. However, it is still a valid indication. E.g. in a power loop, when the arrow moves from bottom to top means that the props passed the vertical and it is time to chop the throttle.

leocb, 2018-07-16:
Whoa, I did not know of that issue, @BlackPixxel is even the guy that made the mod in the video I found randomly on youtube while looking for BF OSD videos. We should definitely get this into the code IMO.
I'll try harder to get in touch with him.

leocb, 2018-07-16:
@Lyonnard Could you post a video or image showing what would be ideal for you? maybe we can work out a pretty darn good Art. Horizon.

Lyonnard, 2018-07-17:
In addition to the current horizon that provides good information when the frame is level I was thinking about something like the following to provide useful information also at high tilt.
The arrow stays on the screen corner and points towards the closest direction to horizon, the [UP] symbol instead moves in the whole area of the screen and indicates up as the current horizon indicates level.
[image]

leocb, 2018-07-17:
Hmm ok, I believe I understand your request, as far as I can tell, the main problem is when fully facing up, the zenith direction is not obvious and can get confusing to orient. Your suggestion is quite easy to do and could complement the current Art. Horizon nicely.
The mod in the video I posted is also a great option, maybe we could have both if enough flash memory is available - to be discussed
I can do this, but it may take a while for me to be able to implement it - I don't have my own quad yet, making things hard to test :(
This would only be ready for 4.0 anyway, so no rush here

BlackPixxel, 2018-07-22:
Hi, here is the already mentioned part of the osd.h that I use in my plane, with the horizon and the arrow pointing to the closest horizon.
https://pastebin.com/1Hy5zTvL

Since I only made it for testing in my plane, and I am not an expert, it is propably very cheaply done.
But if you want to you can of course use the code.
I found especially the arrow to be really usefull when doing loops with the plane, It helps to keep the loop vertical.

mikeller, 2018-07-22:
Thanks @BlackPixxel!

tavdog, 2022-05-27:
Implemented?

haslinghuis, 2022-05-27:
Fixed in: https://github.com/betaflight/betaflight/pull/10417

closed 2022-05-27 by haslinghuis