betaflight/betaflight #6352: BF3.4 cli tasks gives very high maxload

0crap, 2018-07-13:
I went from BF 3.3.3 to 3.4.0 on my Matek F405-STD controller.
When tryping "tasks" on the cli it gives a very high maxload.
avgload seems to be normal.
In the tasks list I see twice the item SYSTEM, one seems to be the cause for the high maxload.

# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       3       1    0.5%    0.5%         0
01 - (         SYSTEM)    983    2135       2  210.3%    0.6%       333
02 - (       GYRO/PID)   7874      58      47   46.1%   37.5%     64671
03 - (            ACC)    988      16       8    2.0%    1.2%      1425
04 - (       ATTITUDE)    492      15       8    1.2%    0.8%       705
05 - (             RX)     32      45      34    0.6%    0.6%       204
06 - (         SERIAL)     99   19763       6  196.1%    0.5%       119
08 - (BATTERY_VOLTAGE)     49       5       2    0.5%    0.5%        16
09 - (BATTERY_CURRENT)     49       3       1    0.5%    0.5%         5
10 - ( BATTERY_ALERTS)      4       3       1    0.5%    0.5%         1
11 - (         BEEPER)     99       6       3    0.5%    0.5%        47
18 - (       LEDSTRIP)    100     128      73    1.7%    1.2%      1341
19 - (            OSD)     59     468      37    3.2%    0.7%       407
21 - (            CMS)     59       4       1    0.5%    0.5%         8
22 - (        VTXCTRL)      4      22       1    0.5%    0.5%         2
23 - (        CAMCTRL)      5       2       0    0.5%    0.0%         0
25 - (    ADCINTERNAL)      0       4       2    0.5%    0.5%         0
26 - (       PINIOBOX)     19       3       1    0.5%    0.5%         2
RX Check Function                   4       2                         9
Total (excluding SERIAL)                       270.1%   47.1%

etracer65, 2018-07-13:
This is just a reporting artifact. The "max" columns capture the highest value seen and during initialization there are a lot of things happening and a task might get delayed and capture the false high value.  A change to make the reporting better has already been merged.

https://github.com/betaflight/betaflight/pull/6297

0crap, 2018-07-13:
Great!

I tested leaving the quad on for 4 minutes and it keeps giving very high maxload.
So it seems here not only after initialization.

However, if you think this is the same issue you are referring to, please close this issue.

etracer65, 2018-07-13:
If it continues to be the same high maxload number (ie. not changing) then yes, it's just the non-representative value captured during initialization.

But you can test it out yourself. Grab the latest dev build and see: https://ci.betaflight.tech/job/Betaflight/lastStableBuild/artifact/obj/

I would expect you'll see a high value on the first `tasks` display (showing the initialization), but subsequent iterations will show only the highest value since the previous command.