betaflight/betaflight #6357: GPS altitude not steady in OSD, but displayed OK in Configurator

aowi7280, 2018-07-14:
**Describe the bug**
GPS altitude is not steady in OSD display. It bounces around 1500' or more. It is OK in Configurator, <1meter variation.

**To Reproduce**
Power up the quad. Wait for GPS to locate 8+ satellites. I have 14-16 locked.

**Expected behavior**
Altitude should be reasonably steady.

**Flight controller configuration**
'''# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.4.0 Jul  8 2018 / 04:24:36 (cce8a59be) MSP API: 1.39

board_name OMNIBUSF4SD
manufacturer_id 

# name
name Androne

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature RSSI_ADC

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 1 0 900 1200 0
aux 1 2 0 1300 1700 0
aux 2 10 2 1800 2100 0
aux 3 11 2 1300 1700 0
aux 4 46 3 1800 2100 0
aux 5 13 3 1300 2100 0
aux 6 26 1 900 2100 0
aux 7 35 1 900 1200 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h
rxfail 4 s 1000
rxfail 5 s 2000
rxfail 6 s 1000
rxfail 7 s 1500

# master
set gyro_calib_duration = 175
set acc_calibration = -8,122,-97
set mag_declination = -8160
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set min_throttle = 1050
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 25
set failsafe_off_delay = 160
set failsafe_throttle = 1485
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set bat_capacity = 2200
set ibata_scale = 470
set small_angle = 180
set auto_disarm_delay = 30
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 150
set gps_rescue_ground_speed = 1000
set gps_rescue_throttle_p = 100
set gps_rescue_throttle_min = 1400
set gps_rescue_throttle_max = 1750
set gps_rescue_throttle_hover = 1525
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_min_sats = 7
set deadband = 4
set yaw_deadband = 5
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_alt_alarm = 375
set osd_vbat_pos = 2401
set osd_rssi_pos = 2106
set osd_throttle_pos = 2241
set osd_crosshairs_pos = 200
set osd_ah_sbar_pos = 200
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2456
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 2099
set osd_gps_lon_pos = 2112
set osd_gps_lat_pos = 2129
set osd_gps_sats_pos = 2080
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2094
set osd_compass_bar_pos = 2442
set osd_altitude_pos = 2262
set osd_warnings_pos = 2409
set osd_pit_ang_pos = 2273
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass2_hz = 175
set dterm_notch_cutoff = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 80
set iterm_relax = RP
set iterm_relax_cutoff = 10
set throttle_boost = 8
set p_pitch = 63
set i_pitch = 45
set d_pitch = 38
set p_roll = 55
set i_roll = 40
set d_roll = 33
set p_level = 35
set i_level = 33
set d_level = 50

# rateprofile
rateprofile 0

set roll_rc_rate = 110
set pitch_rc_rate = 110
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 73
set pitch_srate = 73
set tpa_rate = 15'''

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; RTFQ Omnibus F4 Pro
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
OMNIBUSF4SD (OBSD) 3.4.0 Jul  8 2018 / 04:24:36 (cce8a59be) MSP API: 1.39
ESC: Robot Dog BlHeli_32
Motors: SunnySky 2207, 2100KV
Frame: Alien 6" w/5544 props
Receiver FrSky X4R-SB
Vtx: ImmersionRC 600mW 5.8G 5.8GHz
GPS: Holybro Micro M8N
 - how are the different components wired up.
Camera to video In on FC, video Out to VTX
**Additional context**
Add any other context about the problem here.

mikeller, 2018-07-14:
Are you sure it's bouncing around, and not just displaying absolute altitude, like it is meant to do before the first time it's armed?

aowi7280, 2018-07-14:
I am sure. The OSD reads 3200', then 4000', then 2500', etc. when disarmed. I am at ~5000 feet elevation. The numbers are flashing also, blinking on and off. When I arm, it goes to about zero like expected. During flight, altitude seems normal and reasonably accurate. When I am sitting at my computer attached through Configurator, it reads a steady 152.x meters +/-1 meter, but the radio OSD number moves around. I know about the x10 discrepancy so I am not worried about that for now. 1520 meters is about right when converted to feet. It is the OSD number that moves around a lot.

aowi7280, 2018-08-13:
Still a problem on bf3.5rc2

aowi7280, 2018-09-12:
Still an issue in bf3.5.0

closed 2018-10-19 by stale[bot]