betaflight/betaflight #6360: F3 boards lua script not working after 3.4.0 update.

dmagicone, 2018-07-15:
**Describe the bug**
I have 2 SPRacing F3 boards which I updated to 3.4.0 and after the update, I am no longer able to get the lua script to show any values. I still have 1 SPRacing board on 3.3.0 which the lua script works fine. The lua script also works fine on 3 different F4 board I have so the problem seems to be limited to F3 boards or just the SPRacing F3.

**To Reproduce**
Attempting to use lua scripts with the SPRacing F3 shows no values nor gives any error.

**Expected behavior**
The lua script should show values, as it did when using betaflight 3.3.0

**Flight controller configuration**
``` # Betaflight / SPRACINGF3 (SRF3) 3.4.0 Jul  8 2018 / 04:32:14 (cce8a59be) MSP API: 1.39

board_name SPRACINGF3
manufacturer_id 
name 

feature -RX_PPM
feature RX_SERIAL
feature SOFTSERIAL
beeper -SYSTEM_INIT
beeper -ON_USB
map TAER1234
serial 2 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
aux 0 0 0 1125 2100 0
aux 1 1 1 1900 2100 0
aux 2 13 2 1900 2100 0
aux 3 28 1 900 1200 0
aux 4 35 5 1975 2050 0
set gyro_sync_denom = 2
set acc_calibration = -89,-10,-241
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_device = NONE
set min_throttle = 1082
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set yaw_deadband = 4
profile 0

set p_level = 30
set i_level = 30
set d_level = 70
rateprofile 0

set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 10
set roll_srate = 83
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 12
```

**Setup / Versions**
 - Flight controller [SPRacing F3];
 - Other components [frsky x4r akk vtx];
 - SBUS and soft serial for telemetry

**Additional context**
Telemetry works fine just nothing from lua script.

mikeller, 2018-07-15:
@dmagicone: Try this to see if it helps: 
[betaflight_3.5.0_SPRACINGF3.zip](https://github.com/betaflight/betaflight/files/2195409/betaflight_3.5.0_SPRACINGF3.zip)

dmagicone, 2018-07-15:
Still no values.

mikeller, 2018-07-15:
Looking into putting #6098 into 3.4.1 to address these issues with a fix for FPort.

dmagicone, 2018-07-15:
After seeing someone elses committ about 3.4.0 RC5 working and 3.4.0 not working I gave RC5 a try and lua script does work in RC5 for me. I am using sbus and softserial not fport.

mikeller, 2018-07-16:
Try this @dmagicone:

[betaflight_3.5.0_SPRACINGF3.hex.zip](https://github.com/betaflight/betaflight/files/2197645/betaflight_3.5.0_SPRACINGF3.hex.zip)

dmagicone, 2018-07-16:
Worked perfectly and fast. Thanks.

kako0sb, 2018-07-17:
@dmagicone u try 3.5 or use the rc5?

dmagicone, 2018-07-18:
@kako0sb I used the betaflight_3.5.0_SPRACINGF3.hex posted above.

closed 2018-08-24 by stale[bot]

bckenneth, 2019-02-26:
I have the same issue on SPF3 Evo,but on 3.5.5 any fix?