betaflight/betaflight #6369: FrSky Smartport telemetry (So also LUA scripts) unstable after upgrade to Betaflight 3.4

Maarten112, 2018-07-15:
**Describe the bug**
After upgrading my X-Racer F303 board with the X_RACERSPI Betaflight 3.4 firmware Smartport telemetry is very unstable. This results in my FrSky Taranis transmitter shouting at random times a "sensor lost" message. 

Because of the unstable SmartPort telemetry output the Taranis LUA scripts to configure betaflight also behave eratic. Sometimes values are shown, sometimes not. Sometimes it is possible to update a value, sometimes not, etc. 

**To Reproduce**
Upgrade a X-Racer F303 board with the X_RACERSPI Betaflight 3.4 firmware and enable SmartPort Telemetry. 

**Expected behavior**
A stable SmartPort Telemetry output

**Flight controller configuration**
'''
# Betaflight / X_RACERSPI (XRC3) 3.4.0 Jul  8 2018 / 04:37:12 (cce8a59be) MSP API: 1.39

board_name X_RACERSPI
manufacturer_id 
name MartianII

feature -RX_PPM
feature RX_SERIAL
feature TELEMETRY
beeper -ON_USB
map TAER1234
serial 1 32 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 0 1300 2100 0
aux 1 27 3 1400 2100 0
aux 2 26 2 1300 1700 0
aux 3 28 0 1700 2100 0
set acc_hardware = NONE
set acc_calibration = 8,46,-243
set min_check = 1020
set serialrx_provider = SBUS
set min_throttle = 1100
set motor_pwm_protocol = MULTISHOT
set failsafe_switch_mode = KILL
set small_angle = 180
set pid_process_denom = 1
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 48
set i_pitch = 57
set p_roll = 44
set i_roll = 57
rateprofile 0

set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 73
set tpa_rate = 25
set tpa_breakpoint = 1600
'''

**Setup / Versions**
 - X-Racer F303
 - FrSky XSR receiver
 
**Additional context**
SmartPort telemetry (and LUA scripts on Taranis) worked great before upgrading to Betaflight 3.4

mikeller, 2018-07-15:
Looking into putting #6098 into 3.4.1 to address these issues with a fix for FPort.

mikeller, 2018-07-16:
Try this, @Maarten112:

[betaflight_3.5.0_X_RACERSPI.hex.zip](https://github.com/betaflight/betaflight/files/2197670/betaflight_3.5.0_X_RACERSPI.hex.zip)

m013, 2018-07-16:
i was recommended to try fport as well.
i cant apply any settings via luascripts anymore via sbus/smartport(softserial) - the process stucks at 'retrying '

mikeller, 2018-07-17:
@m013: That is with the changes from #6382?

mikeller, 2018-07-17:
@m013: Again, have you tested with the changes from #6382?

Maarten112, 2018-07-17:
@mikeller betaflight_3.5.0_X_RACERSPI.hex (Removed delay from SmartPort frame processing) fixed the FrSky Smart Port issues! Thanks!

closed 2018-07-17 by Maarten112