betaflight/betaflight #6371: Camera Control Menu Opening During Flight

seaweeduk, 2018-07-15:
The camera control menu opened for me during a flight today. I was at around mid throttle and full yaw right when it happened. Seems there is no check that the quad is armed? 

Flight controller was a betaflight F3 with camera control wired to the LED port.

DieHertz, 2018-07-15:
https://github.com/betaflight/betaflight/blob/21414faf1ce9b4f28216ee17b1674e38d6e5158b/src/main/fc/rc_controls.c#L296

DieHertz, 2018-07-15:
if (ARMING_FLAG(ARMED) || doNotRepeat || rcDelayMs <= STICK_DELAY_MS || (getArmingDisableFlags() & ARMING_DISABLED_RUNAWAY_TAKEOFF)) {
        return;
    }

DieHertz, 2018-07-15:
https://github.com/betaflight/betaflight/blob/21414faf1ce9b4f28216ee17b1674e38d6e5158b/src/main/fc/rc_controls.c#L231-L233

seaweeduk, 2018-07-15:
Sounds like this is something shorting on my side

closed 2018-07-15 by seaweeduk