betaflight/betaflight #6372: TBS Unify Pro Nano always starts in pit mode

andrejpodzimek, 2018-07-16:
I'm using a [HGLRC F4 Zeus](https://www.hglrc.com/products/hglrc-f4-zeus-f4-flight-controller-integrated-with-osd-bec-pdb-aio-15a-blheli_s-4-in-1-esc-for-rc-racing-drone) with [TBS Unify Pro Nano](http://www.team-blacksheep.com/products/prod:tbs_unifypro_nano). After finally stumbling upon a Nano that doesn't immediately fail **(*)** (with a few lessons learned in the process), I'm struggling with automatic pit mode on the Nano.

Right after boot the video is on the right frequency configured in Betaflight, but moving my receiver ~3 meters away from the drone makes the signal deteriorate into randomness. This makes me assume that the VTX is in pit mode.

To get out of pit mode, one has to either hold the VTX button (which I soldered on, fortunately) for ~3 seconds or go to the OSD menu and configure `FREE` mode instead of `RACE` mode deep in the `SA VTX` submenu. Once I do that, the VTX resets itself (the signal vanishes for a short moment) and then I get a way better signal with the full 25 mW of power.

Because I don't fly FPV and use the VTX only for (analog) camera recording, an automatic pit mode that can only be deactivated through the OSD menu is somewhat unpleasant. :disappointed:

Would it be possible to make the `FREE` / `RACE` switching automatic? Or does this have something to do with channels? I'm using F5; just picked a channel at random. Are there any channels that would cause the VTX to **not** boot in pit mode?

For the record, here's my `diff all`:

```shell
# diff all

# version
# Betaflight / OMNIBUSF4 (OBF4) 3.3.3 May 31 2018 / 13:23:55 (3565182d3) MSP API: 1.38

# reset configuration to default settings
defaults nosave

# name
name Windy

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beeper

# map

# serial
serial 2 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led
led 0 6,0:D:ATOIW:0
led 1 7,0:D:ATOIW:0
led 2 8,0:D:ATOIW:0
led 3 9,0:D:ATOIW:0

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 13 3 1300 2100 0
aux 4 28 4 1700 2100 0

# adjrange
adjrange 0 0 2 900 2100 12 2

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_p_ratio = 0
set blackbox_device = NONE
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set current_meter = NONE
set pid_process_denom = 1
set osd_warnings = 63
set osd_vbat_pos = 2466
set osd_rssi_pos = 2434
set osd_flymode_pos = 2476
set osd_throttle_pos = 2485
set osd_vtx_channel_pos = 2452
set osd_crosshairs_pos = 2282
set osd_ah_sbar_pos = 2282
set osd_ah_pos = 2282
set osd_current_pos = 427
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_channel = 5
set vtx_freq = 5820
set vcd_video_system = 1

# profile
profile 0

set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 80

# profile
profile 1

set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 80

# profile
profile 2

set vbat_pid_gain = ON
set anti_gravity_gain = 3000
set dterm_setpoint_weight = 100
set d_level = 80

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# rateprofile
rateprofile 2

set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 80

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

**(*)** Ad. TBS Nano: From my recent experience, ~1 out of 3 Nanos works while the other 2 overheat themselves to death within a few minutes, stop transmitting, stop responding to SmartAudio and that's the end of their story. My recommendation: First, never use a Nano in a heatshrink. It will be gone within a minute. Second, never use a Nano without [the 20x20 mm original mounting board](http://team-blacksheep.com/products/prod:unify_xf_mountingboa), which provides extra cooling capacity. Without the mounting board, it will be gone in ~5 minutes on the bench.

andrejpodzimek, 2018-07-19:
So if I power-cycle it later (rather than immediately), the mode won't get saved?

The problem I'm trying to solve is that I don't want to use any OSD menus to switch to the free mode, because I don't use FPV. I just want to plug in the battery and be reasonably sure that my analog signal (for recording) is available.

jflyper, 2018-07-19:
> So if I power-cycle it later (rather than immediately), the mode won't get saved?

I think so.

> The problem I'm trying to solve is that I don't want to use any OSD menus to switch to the free mode, because I don't use FPV. I just want to plug in the battery and be reasonably sure that my analog signal (for recording) is available.

Don't use SmartAudio then.

andrejpodzimek, 2018-07-19:
> Don't use SmartAudio then.

Huh? Why? It's way easier to set the VTX channel in Betaflight's CLI than using the weird little button&hellip; And I must have SmartAudio for that to work.

andrejpodzimek, 2018-07-19:
[Most of it is already implemented, right?](https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings) Because the CLI channel settings just work, without using the OSD menu. So the only small missing bit would be the `FREE` or `RACE` mode. But if it can be "stored" on the VTX by power-cycling it quickly after the mode switch, then I guess there's not much to implement.

Is there anything wrong about the current "authority of configuration" that would require a PR? I don't think so&hellip; IIUC, the Betaflight CLI can be used to specify the VTX settings and unless I go into the OSD menu and change something, my CLI settings are the defaults. That looks perfectly fine to me.

Does it make sense to implement the `RACE` / `FREE` switching as another CLI variable? If not, then let's just close this.

closed 2018-07-19 by jflyper