betaflight/betaflight #6373: Per-ESC Telemetry Confuses Opentx

WarriorOfWire, 2018-07-16:
**Describe the bug**
There are 5 jumbled up voltage and current sensors in Opentx reported by Betaflight through an S.Port to a Taranis x7s.

**To Reproduce**
Steps to reproduce the behavior:
* Enable ESC voltage and current sensors.
* "Detect sensors" in Opentx.
* Observe many redundant sensors.

**Expected behavior**
Well-labeled _or_ single voltage/current sensors are sent over S.Port for transmitter telemetry.

**Flight controller configuration**
```
# version
# Betaflight / FPVM_BETAFLIGHTF7 (FBF7) 3.4.0 Jul  8 2018 / 04:12:22 (cce8a59be) MSP API: 1.39

board_name FPVM_BETAFLIGHTF7
manufacturer_id 

# name
name Quad

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 1024 115200 57600 0 115200
serial 1 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2050 0
aux 1 1 1 950 1025 0
aux 2 2 1 1475 1525 0
aux 3 28 1 1975 2050 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1500

# master
set acc_calibration = -36,-73,-359
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = AUTO-LAND
set bat_capacity = 2000
set vbat_min_cell_voltage = 34
set current_meter = ESC
set battery_meter = ESC
set vbat_divider = 1
set small_angle = 45
set yaw_deadband = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 2450
set osd_alt_alarm = 111
set osd_vbat_pos = 2464
set osd_rssi_pos = 2432
set osd_remaining_time_estimate_pos = 2546
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2528
set osd_gps_lon_pos = 233
set osd_pid_roll_pos = 486
set osd_battery_usage_pos = 2533
set osd_disarmed_pos = 2474
set osd_esc_tmp_pos = 2552
set osd_esc_rpm_pos = 2520
set osd_core_temp_pos = 2488
set displayport_max7456_blk = 2

# profile
profile 2

# rateprofile
rateprofile 1
```

**Setup / Versions**
 - Flight controller: FBF7 board
 - FrSky XSR transciever
 - No video yet
 - All soldered to board

**Additional context**
I'm new to the scene; I can't figure out which telemetry line is the aggregate one, and the per-ESC measurements are not useful while costing valuable telemetry sensor slots.  A mechanism to disable reporting the per-ESC telemetry to the transmitter would be ideal.

roeemz, 2018-07-16:
I have noticed the same thing, ESC telemetry is useless when you have current sensing in the fc and I just turned it off because it is spamming my Taranis. 

S. Port is having issues anyway, this doesn’t help

WarriorOfWire, 2018-07-16:
My fc's voltage and current sensing does not work correctly with Betaflight.  The ESC telemetry is my only recourse, and tbh it's probably more accurate than the FC anyway.

closed 2018-08-30 by stale[bot]