betaflight/betaflight #6376: GPS Rescue - Copter disarmed in air with RESCUE_SANITY_ON

Quickred, 2018-07-16:
To detect the landing, I change in the CLI
set gps_rescue_sanity_checks = RESCUE_SANITY_ON

If GPS rescue is activated, the copter disarmed, if the  gps_rescue_descent_dist is reached immediatly.

Video here:
https://youtu.be/fFLLa-1T0cA?t=1m

Should there anything change in the GUI? Stage 2 in failsafe to landing instead of drop?

pmijoiu, 2018-07-17:
I have the same issue. If sanity check is Off it arrives back home without any issues.

With sanity check On it disarms at 96m from home. Did it 2 times at exactly 96m.

Here are my settings:
gps_rescue_initial_alt = 20
gps_rescue_ground_speed = 1000
gps_rescue_angle = 32(default)
gps_rescue_descent_dist = 60
gps_rescue_sanity_checks = RESCUE_SANITY_ON

Here are the video that I've posted in Long range hooligans (facebook group):

https://www.facebook.com/paul.mijoiu/videos/pcb.1910882879006093/1977686852283673/?type=3&theater&ifg=1

https://www.facebook.com/paul.mijoiu/videos/pcb.1910882879006093/1977687738950251/?type=3&theater&ifg=1

s0up, 2018-07-17:
https://github.com/betaflight/betaflight/pull/6392

callagga, 2018-08-05:
seems like it's included in release 3.4.1 (https://github.com/betaflight/betaflight/releases)..

@mikeller This would be also fixed in 3.5.0 RC2 I assume?   i.e. looks like both these releases came out just days ago around the same time, so just checking does the higher version RC release automatically have all the fixes from the lower version (3.4.1)?

siamect, 2018-08-05:
3.5.0 RC2 contain the changes  "fix the logic to be a little bit more user friendly"  b69e97fb23b2e and the "fix disarm logic" 4b59726 but I still have the problem so i guess there is something in the performSanityChecks() that still kicks in...

callagga, 2018-08-22:
did it get fixed in the final version of 3.5?

Diaoul, 2018-09-04:
I'm having the same issue, the quad disarms when I flip the GPS RESCUE switch. The quad is 130m away with `gps_rescue_descent_dist = 60` and `gps_rescue_sanity_checks = RESCUE_SANITY_ON`.

It's not clear to me what `gps_rescue_sanity_checks` does and I have no idea why it drops out of the sky. Is there a way to see what's going on? Maybe on OSD something I missed?

siamect, 2018-09-04:
The sanity check doing stuff like crash detection, flyaway detection, stalled movement detection, minimum satelltie detection and I think that one of these goes wrong so...
 
I just disabled the sanity and went into the sources and changed the
100m to 25m line 120 something in the gps_rescue.c
So  it looks like this now:  

// Minimum distance detection (100m).  Disarm regardless of sanity check configuration.  Rescue too close is never a good idea.  
         if (rescueState.sensor.distanceToHome < 25) {  
  
But a big but... You trade safety for convienince because it is almost
always safer to drop out of the sky (provided you fly over safe
grounds) than to let your drone make it's own thing, especially with
the sanity off. It will probably burn up if it hits anything on the way
home or if you have anything failing that causes the quad to crash.  
  
So be careful and take care.  
Martin

closed 2018-10-11 by stale[bot]