betaflight/betaflight #6384: BF 3.4 - Problem with Loop PID 8K and DSHOT600

brunoleaomaia, 2018-07-16:
Hi All, 

I had a problem running PID Loop in 8 Khz and DSHOT600. Hovering in angle mode (throttle in ~25%), I had many motor glitches, a really fast stop and start, radomnamically between the motors.

It's a new build, all parts are brand new.

- FC Matek F405 CTR (BF 3.4.0)
- ESC DYS XSD 30A BLHeli_S (v.16.6)

I changed the PID Loop to 4 Khz and is running OK until now.

chinoox, 2018-07-16:
What is the cpu load in betaflight configuration?

brunoleaomaia, 2018-07-17:
@chinoox the CPU load is 8% with pid loop at 4 Khz and 10% with pid loop at 8 Khz.

atomiclama, 2018-07-17:
It's a problem with your ESC not betaflight.

Some ESCs can't handle Dshot properly, they will turn the motor but you might get ticks and glitches.
Drop the protocol to Oneshot 42 and you should be fine.

Oneshot has less processor overhead than Dshot.

tonino1991, 2018-07-19:
did you connect the signal ground to your fc??? sometimes that does the trick

closed 2018-08-25 by stale[bot]