betaflight/betaflight #6386: RevoltOSd arming problem

rcquad808, 2018-07-16:
Need some help. Running BF 3.4.0 on revoltosd, Blheli32 dsot600. Everything is working in BF configurator, Motors all spin, radio/xsr receiver all working. When I arm (usb not connected to the fc) it does not arm. I even tried a new receiver and reflashed the fc couple times and still not arming. I flashed it with RF configurator with the lastest RF firmware and it arms when i flip the arming swtich.

etracer65, 2018-07-17:
Assuming you have the "Warnings" OSD element enabled (it is by default) and the "Arming Disabled" warning item enabled (also on by default), then you should see the reason it won't arm displayed in your OSD.

Also are you sure you properly set up your arming switch on the "Modes" tab?

rcquad808, 2018-07-17:
@etracer65
Yes everything is default setting and everything in the “Modes” tab is all correct. I reflashed it and it armed correctly, but when I unplug and replug the battery it didn’t arm and still not arming. I got an arming code of 3,17 but I checked those and has nothing to do with it. Both checked fine.

etracer65, 2018-07-17:
Well arming disabled code 3 is `RX LOSS` so that's a problem.  Have you confirmed that your receiver is getting power when powered by the battery and USB is disconnected?  If so, have you looked at the OSD when trying to arm to see if any additional reasons are listed as I described above?

closed 2018-08-24 by stale[bot]