betaflight/betaflight #6387: Omnibus F7V2 BF3.4.0

Ralfde, 2018-07-17:
Gyro and accels do not work very often. Need to reflash the FC incl. bootloader to get it work again. Other users seem to have similar problems with the Omnibus F7V2.

jflyper, 2018-07-17:
Are you saying that working acc/gyro spontaneously stop working, and it is necessary to re-flash to make them work again?

Ralfde, 2018-07-17:
correct, sometimes it stops working completely, sometimes reaction on the PC looks strange and slow and not showing the correct position of the copter.

Ralfde, 2018-07-18:
I was not using the F7 before. I used the F4V5 board before. Now, as BF 3.4.0 promise much better flight performance for F7 FCs, I tried it now. But I believe 3.3 was okay, because I did not read and hear about any trouble with this.

jflyper, 2018-07-18:
@Ralfde Can you try v3.3.3 for basic acc/gyro tests?

Ralfde, 2018-07-18:
I can try, but as I do not know, which change in settings does deactive the gyro it maybe does not make sure about it. But I can do some tests.

Ralfde, 2018-07-18:
3.3.2 seem to make already trouble, too. Updating was not possible now and after trying this the gyro did not work, too. I had to update with pushed button in dfu mode again with complete bootloader.
3.3.1 now shows also strange values. Flashing 3.3.0 did show also strange values, flashing again 3.3.0 solved the problem vor now. I believe something is overwritten. After changing from 32kHz gyro to 8kHz gyro, even with 3.3.0 the gyro does not work. After flashing the firmware again (without bootloader this time), the gyro worked with 8kHz. I'm afraid the F7V2 was never bug free.

Ralfde, 2018-07-18:
Omnibus F7NXT does not look much better. The 32kHz Gyro setting does not work. The 8kHz Gyro setting seem to work a little more stable.

jflyper, 2018-07-18:
@Ralfde 
Weird. Omnibus F7 V2 and OmniNXT7 I have for testing works stably with v3.4.0, 32/32.

What do you mean by "I had to reflash the FC"? You could only restore the control over the board via BFC only by re-flashing, because the board became non-responding by a particular setting?

Ralfde, 2018-07-18:
I just changed to the 32kHz gyro frequency, save and reboot.

jflyper, 2018-07-18:
What made you force re-flashing?

Ralfde, 2018-07-18:
One more thing I tested. The magnetometer IST8310 with I2C address 0x0C is not found, yet.

Ralfde, 2018-07-18:
What made you force re-flashing?
Because the gyro reaction was not shown normally in the configurator. The gyro did not work.

jflyper, 2018-07-18:
Did it detect the gyro?

Ralfde, 2018-07-18:
Which firmware did you use now? 3.4.0 or 3.3.0?

jflyper, 2018-07-18:
F7 only works stably with v3.4.0.

Ralfde, 2018-07-18:
Gyro is yellow, but the copter does not move correct:
[image]
The F7NXT seem to work at least stable until now with 8/4khz. But the F7V2 does not work. (works with inav 2.0.0 RC4, so it should be no hardware problem.

jflyper, 2018-07-18:
Let's put NXT7 aside for now. It's confusing to go after two different things.

So, how not correct is "does not move correct"? Can you describe? Movie clip would be the best... Also, do you know which gyro you are using right now?

Ralfde, 2018-07-18:
Most time the copter on the PC does just not move or sometimes it makes it own movements slowly.
Now, when the copter did not move, I calibrated the accelerometers and after this, copter did move. But after some time it's turning slowly itself again.

One question: What else do I need to change to set it to 32kHz correctly?
Just set it like this:
[image]
But after save and reboot it does show this: 32/2kHz
[image]

Ralfde, 2018-07-18:
Tried now another board. Gyro seem to work fine now. But very strange, that the other board works fine with inav. I think some tolerances are maybe too tight. If you want I can send you this board with the problems.
Anyway, please tell me, what else I need to do to make the gyro work correctly with 32/32.

Ralfde, 2018-07-18:
With the bad F7V2, the accel values make strange things (copter not moving all the time:
[image]

Ralfde, 2018-07-18:
[image]

jflyper, 2018-07-18:
Bad gyro or gyro FPC. Talk to your distributor.

Ralfde, 2018-07-18:
ok. Really looks like a bad gyro/accel. Thanks a lot.

Anyway, please let me know how to set 32/32 correctly and how to use the IST8310 with 0x0C address.

jflyper, 2018-07-18:
IST8310? How did #5386 end up?

Ralfde, 2018-07-18:
I did get a "hack" firmware, which detected the IST8310, but since this time nothing more happend, I believe.

Ralfde, 2018-07-18:
Very strange. The problem hardware seem to work without any problems on inav.
[image]

jflyper, 2018-07-18:
Do cli `status` and see which gyro it's using.

Ralfde, 2018-07-18:
So, 32/32 setting works, if DShot is set to 1200. But CPU load is 100%. So I went back to 16kHz PID loop now. Are there somewhere advised settings for the 32kHz gyro mode? Didn't find anything.

Ralfde, 2018-07-18:
I compared now the gyros. The 32kHz gyro seem to have much more vibrations compare to the MPU6000. Don't think this makes a lot of sense, if there are not magic filters which could make the 32kHz gyro better as the MPU6000.

closed 2018-07-18 by Ralfde