betaflight/betaflight #6400: Smartaudio via softserial for working on omnibusf4 board

AlessandroAU, 2018-07-19:
Smartaudio via softserial for working on omnibusf4 board

Tried mapping softserial1 TX to A01 and B12, neither works, on the scope the voltage was steady at around 2v for A01 and 2.8v using B12, there was a minor dip to 0v during startup. Otherwise no activity ever even when changing vtx settings

Remapped to hardware uart 6 and it is working fine.

running release firmware 3.4

FredFrancisIT, 2018-07-21:
Same issue here... cannot get feature -RX_SERIAL activated

AlessandroAU, 2018-07-21:
I'm using a cheap clone board, https://www.banggood.com/Micro-20x20mm-Betaflight-Omnibus-STM32F4-F4-Flight-Controller-Built-in-BEC-OSD-for-RC-FPV-Racing-Drone-p-1132495.html

I would be inclined to believe my A01 pin may be faulty but there still still not activity when I direct soldered and remapped to PB12

jflyper, 2018-07-22:
@FredFrancisIT

Your issue is something completely different.

jflyper, 2018-07-22:
@AlessandroAU

Did you solder directly to MCU for PB12?

jflyper, 2018-07-22:
There’s a chance that PB12 is occupied with something else ...

AlessandroAU, 2018-07-23:
Yep, I have directly soldered to PB12 as it was not broken out.

jflyper, 2018-07-24:
PB12 may be in use, if the board is a clone of one of nano variants ...

closed 2018-08-30 by stale[bot]