betaflight/betaflight #6401: [Feature Request] Change I2C speed and timout from CLI

McGiverGim, 2018-07-19:
Betaflight uses a fast I2C clock (800kHz), the double than the standard I2C bus uses as maximum.

This is needed to not wait too much for I2C readings because Betaflight is focused in performance. 

This is one of the reasons because it seems some I2C compass did not work. Some of them work when Betaflight I2C bus works at 400kHz and the I2C timeout is changed from SHORT to LONG.

I think it will be good to have this two parameters configurable. In this way some compass that now did not work will start to work. It must be an CLI param because only users with problems must test this, if you are focused in performance is better to not change it at all.

This is not a solution for every compass, and I'm sure that the principal problem is at a more low level than that, but is a workaround for some compass until someone with the material, the resources and knowledge can fix it for all.

The I2C speed is now part of a PG, but I'm not able to figure how to publish it in CLI because the number of elements is variable depending of some defines.

These are the PG defines for the I2C bus:
https://github.com/betaflight/betaflight/blob/f2468fb894129eefad9ae9c53e4bbb67b15015ed/src/main/pg/bus_i2c.h#L36

And the element that I want to change from CLI is the overClock field:
https://github.com/betaflight/betaflight/blob/f2468fb894129eefad9ae9c53e4bbb67b15015ed/src/main/pg/bus_i2c.h#L32

And the timeout that I want to expose in CLI too is now a define:
https://github.com/betaflight/betaflight/blob/f2468fb894129eefad9ae9c53e4bbb67b15015ed/src/main/drivers/bus_i2c_impl.h#L28-L30

If someone can do this or show me how to do it will be great. Now I'm doing a custom firmware for my quad, but CLI params are always better.

etracer65, 2018-07-19:
This is somewhat complicated because i2c config is actually a variable length array depending on how many i2c buses are defined for the target.

Also I would assume you'd want the ability to set the overclock and timeout on a per-bus basis rather than for all configured buses.

Since it's an array of struct's rather than an array value inside a struct we can't use the array type CLI parameter.  So short of completely restructuring the PG the only way I can see to do this would be to have conditionally enabled `ic2_bus_X_overclock`  type parameters where `X` is the bus number.

McGiverGim, 2018-07-19:
No necessary. It is a workaround, so if it changes for ALL the buses is enough for the moment. Now is how I'm using it. I have changed them for all the buses and it works.

etracer65, 2018-07-19:
Also this may be a bug, but it looks like for F3 the timeout is hardcoded in drivers/bus_i2c_stm32f30x.c to the long timeout `I2C_LONG_TIMEOUT` rather than using the default `I2C_DEFAULT_TIMEOUT` which would normally be the short timeout.  For F1 and HAL it seems to be using the default short setting.  I don't know if this is a bug or some limitation of the F3.

andrejpodzimek, 2018-07-19:
Just for the record, I'd like to link to #4531 and [this comment in particular](https://github.com/betaflight/betaflight/issues/4531#issuecomment-394095044). That's where some I2C flexibility would be very useful. Otherwise I'll need to patch and rebuild Betaflight for that particular drone forever. :smile:

IIUC, Betaflight uses somewhat aggressive I2C settings to reduce delays to the control loop caused by I2C, caused by the simple fact that [I2C isn't handled asynchronously](https://github.com/iNavFlight/inav/issues/1931#issuecomment-322151866). iNav has far less aggressive I2C settings (which is what #4531 is mosty about), but it doesn't provide any of the crazy fast loop frequencies.

McGiverGim, 2018-07-19:
Yes, that is the first part of changes needed that fixes some mag. Yours needed one change more if I remember correctly and my idea was to study it too, but once the others were merged.
I've the same problem then you, I need to make the changes forever because this is the only gps model not affected by the noise of the runcam split.

andrejpodzimek, 2018-07-19:
<off_topic>Oh yeah, re RC Split noise. That’s just killing me. I have to switch the Split completely off to ever get a GPS hold. Only one odd tiny M8N can get a hold near a Split Mini. Regular Split + large M8N => no way. I don’t fly FPV, so I don’t really mind switching the camera off here and then, but for making videos, the choice of *either* video *or* GPS is super-annoying. You may want to go to the edge of LoS while making videos and that’s exactly where *any* type of GPS input would be nice.</off_topic>

McGiverGim, 2018-07-19:
Sorry for the off topic too, but if it can help you or some other, the radiolink se100 fixes until 20 satellites with the runcam split without problem. Is the only one model that has no problems with it... But the compass need this changes to the i2c to work.

McGiverGim, 2018-07-19:
No, I have it directly in the top plate of my Martian II. I've tested others and is the only model that works. If someone is interested in photos or something can contact with me in slack to not add more off topics here 😁

closed 2023-06-14 by McGiverGim