betaflight/betaflight #6405: kakute F7  motor protocol error

nerilorenzo, 2018-07-20:
**Describe the bug**
kakute F7 board is not able to set properly the motor protocol with betaflight 3.4 and 3.3.3, with 3.3.2 it works

**To Reproduce**
betaflight + kakute f7 + blheli 32

**Expected behavior**
protocol setted correctly

**Flight controller configuration**
after firmware upgrading no changes

**Setup / Versions**
 - kakute f7
 - Tekko32 35A ESC

**Additional context**
it was aploaded a video that explain the problem and also in the comments there are other user comments.
I attached also some pictures I made.

[image]
[image]
[image]

mikeller, 2018-07-20:
@nerilorenzo: Can you actually specify what you expect Betaflight to do, and what does not work? The screenshots are from the BLHeli configurator, and not Betaflight, maybe file an issue there?

nerilorenzo, 2018-07-20:
after loading the betaflight firmware on the card I selected the engine protocol and then tried to test the motors. not all motors have been started. depending on the chosen motor protocol, the number of motors running has changed. I was never able to have all motors running.
I tried with betaflight 3.4 and 3.3 and it doesn't work. In the youtube video there is in a comment saying that with 3.3.2 it works.
to verify the setting of the motor protocol I made a check with the blheli interface.

snajby, 2018-07-20:
Hi, this issue is probably related to this #6306  opened 12 days ago.

nerilorenzo, 2018-07-20:
yes,
what it is better to do? I need to add my files and comments to the other issue page and we close this?

mikeller, 2018-07-21:
Duplicate of #6306.

closed 2018-07-21 by mikeller