betaflight/betaflight #6406: RevoltOSD Arming Issue BF 3.4

umgram1, 2018-07-21:
Running a RevoltOSD on BF 3.4, after fully configured the board won't arm. Arming flags 12 and 17. (Gyro calibration and configurator connection). I read another issue on the same board was fixed by disabling 32 khz mode, but I never had it enabled anyway. 8k and 8k.. I even disabled accelerometer and turned down to dshot600 from dshot1200. Reflash does not solve the issue. Also as a subtext, I build quads for a living. I have another RevoltOSD, but before I try it I will flash flight one on the problem board and see if it arms. If it does arm on Flt1, what could be wrong?

fujin, 2018-07-21:
Increase the `gyro_calib_noise_limit`.

closed 2018-07-21 by fujin

rcquad808, 2018-07-23:
Yes I have the same issue and I did open a thread on it. Also I have been building quads for over 2 years now.   I had it 8K/8K multishot, but it still intermittently comes on and off when arming. I will try to increase the gyro calibration noise limit and see if that will solve the problem.