betaflight/betaflight #6412: Lost propeller detection/compensation. (Auto configuration switch to three motor quad)

NeilGaede, 2018-07-21:
Would it be feasible to implement a feature to detect when/if a prop. spins off and compensate by switching to a three motor configuration?

If the F.C. detects that a motor is no longer delivering thrust, instead of sending it spinning helplessly out of control, can't it compensate by removing the motor from consideration and compensate as if it was designed to only use three motors? 

It might not be pretty or smooth, but at least you could pilot it down and land without totally destroying your quad. 

Thanks for the consideration.

etracer65, 2018-07-21:
Not really possible with a 4 motor setup since how would you have any control in the diagonal axis where the lost motor resided?  Some advanced camera platform flight controllers have the ability to compensate for a lost motor in a 6 or 8 motor configuration since there are still active motors that can provide compensatory thrust.

The only option would be some kind of controlled "spinning frisbee" effect that's been demonstrated academically.  But that wouldn't be controllable from an FPV perspective.

raimokr, 2018-07-24:
Maybe with enabled 3D mode or a kind of? If the motor on the 'degraded' axis could spin in reverse (if the ESCs support that)?

etracer65, 2018-07-24:
Motors can't reverse quick enough for this to be possible.

NeilGaede, 2018-07-25:
What do you mean by diagonal axis? We only have pitch, roll, yaw, and climb/descend. With cg in center of quad, presumably, with only one motor gone, you would still have authority in all axis. Motor on side with dead motor would just have to work harder, and max climb rate would less, obviously, but with power to weight ratio of modern quad being like 500:1 these days I don't think it would be a problem. Would credibility increase if I program my quad for three motors and record it flying? 😉

mikeller, 2018-07-25:
@NeilGaede:

> Would credibility increase if I program my quad for three motors and record it flying? ￼

Yes please, that would be a nice proof of concept.

jflyper, 2018-07-25:
There is no stable 3 motor configuration.

This is obvious as there is exactly a half of the region around the cog with no explicit authority for roll and pitch. For yaw, it is impossible to yaw in one direction, and remaining direction will accompany altitude change.

NeilGaede, 2018-07-25:
OK, didn't work quite so well. I don't think it had quite enough yaw authority. Oh well, back to the drawing board.

mikeller, 2018-07-25:
@NeilGaede: Video? :rofl:

cmot17, 2018-07-29:
https://www.ted.com/talks/raffaello_d_andrea_the_astounding_athletic_power_of_quadcopters

closed 2018-09-05 by stale[bot]