betaflight/betaflight #6413: SpracingF7Dual:  dynamic filter issue when using "both" gyros

flint723, 2018-07-21:
While testing the PR for dynamic filter changes, I noticed that in 3.4.0, 3.5.0 (1008), and the PR, the dynamic filter will not go above 220hz or so at full throttle, despite that there doesn't seem to be any noise spike in `gyro_scaled at 220hz`.  This is verified in a few logs using `debug_mode = fft_freq and gyro_scaled`

If `gyro_to_use = first` is set, the fft_freq will go up where it should (440hz for this quad) at full throttle.  This matches the peak noise in `gyro_scaled` at full throttle as well.

You can also see it in Plasmatree Pid Analyzer noise plots:  No changes between these two images except `gyro_to_use`.  But in the `gyro_to_use = first` image the dynamic filter takes care of the 440hz noise (as expected).  In the `gyro_to_use = both` image, the dynamic filter is just sitting in the middle doing nothing for the higher freq noise.

Gyro to use = both
[image]

Gyro to use = first
[image]

And logs with gyro_scaled and FFt_freq:
[spracing f7 dual gyro to use both.zip](https://github.com/betaflight/betaflight/files/2216472/spracing.f7.dual.gyro.to.use.both.zip)
[spracing f7 dual gyro to use first.zip](https://github.com/betaflight/betaflight/files/2216473/spracing.f7.dual.gyro.to.use.first.zip)
And diff:
[diff 3.5.0 branch dynamic_filter_update, gyro_to_use first.txt](https://github.com/betaflight/betaflight/files/2216479/diff.3.5.0.branch.dynamic_filter_update.gyro_to_use.first.txt)

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

DieHertz, 2018-07-21:
You've found a bug, sort of. Dynamic filter wasn't recommended for anything above 8K, and it's definitely not designed for more than one gyro.

Each gyro is sampled separately and has it's own set of filters. But dynamic filter has global mutable state and is hence not reentrant. What you observe is the filter gets twice the amount of samples in the same time, because both gyro update functions push to the same FFT structures. As a result detected frequency is twice lower than what's expected.

I'll look into making dynamic filter reentrant, not sure how good it will be though.

etracer65, 2018-07-21:
Perhaps we should change the dual gyro logic to average the raw samples _before_ applying the filters and the rest of the gyro logic.  Move the zero offset, slew filter and sensor alignment out of `gyroUpdateSensor()` and perform them for both gyros in `gyroUpdate()` and calculate the average before calling `gyroUpdateSensor()`.  The same could apply to the single gyro logic.  So it this way would let the dynamic filter work with the code the way it is.

DieHertz, 2018-07-21:
Not sure it's worth moving averaging around when we don't know if it will stick with us for long.
We may have something more interesting in the future and not sure whether it would be done before or after other filters.