betaflight/betaflight #6425: FEATURE REQUEST: Disable OSD when in Camera Control Mode

joshuabardwell, 2018-07-22:
OSD gets in the way when camera control is active. Suggest automatically disabling the OSD when camera control becomes active.

etracer65, 2018-07-22:
Not sure exactly what you mean - disable the OSD display or something to do with CMS menus activated from the OSD.

The problem is the same in both cases - we don't really know for sure that camera control is active.  Camera control just emulates a dumb button dongle and there is no feedback from the camera that we're actually in the camera's OSD/settings.

If the goal is to blank the OSD because it interferes with the camera menus and makes them difficult to read, then you could set up the "OSD DISABLE SW" mode to blank the OSD when you want.

ictea-rc, 2018-07-23:
etracer65, actually there is a way to know if in camera control mode: for Mode 2, the Yaw control joystick is turned to the right completely, while all other joysticks are at the center position.

etracer65, 2018-07-23:
@ictea-rc Yes, I'm aware of the stick commands.  But while we can detect that a stick command was sent, we can't actually tell if the camera has actually entered it's settings menu.  All we know is that we sent the "button press".  So what if the camera control wire wasn't connected?  Or if the settings weren't quite right and the menu was never activated?  We have no way of knowing.  So we may blank the OSD (with no way to get it to return) even though the camera menu may or may not be visible.

But actually much worse is that we have no way of knowing that the user has exited the camera menu and we should reactivate the OSD.  Most analog cameras exit their menus by having the user scroll to a "Save" or "Exit" selection.  We have no way to detect that.

ictea-rc, 2018-07-23:
I understand that for joystick control this is not possible. How about the cameras that are controlled using an UART (TX and RX)? I'm assuming that in that case the "In and Out of camera control are sent by the FC and are well defined and known.  The number of cameras with UART control is growing, not to mention the RunCam control adapter board.

ictea-rc, 2018-07-23:
Just to be clear: I'm talking about RunCams that are controlled using the TX/RX function of one UART (universal asynchronous receiver-transmitter). The UART is essentially a digital device. It sends and receives digital information not analog. 
[image]
The Micro Eagle above can be bought with such a control.  In this case, the camera is not as dumb as a analog security camera (by now pretty much everybody knows that older FPV cameras are re-branded security cameras), it has a controller that interfaces with the TX/RX of the OSD control. Keep in mind that cameras that have OSD via TX/RX have no external joystick for camera settings. All is done inside the camera and controlled by the FC, In the case of OSD control via TX/RX, the flight controller initiates the transaction based on the position of the joystick and because it has to open and close the UART I'm assuming the FC also knows when to end it. I'm not the designer that implemented that interface, but it seems that the FC should be in control of the operation of setting the camera. The board sold by RunCam to interface a camera with old OSD functionality translates commands from the FC to the camera that understands joystick (analog) commands, that's all.

mikeller, 2018-07-23:
@ictea-rc: Please read https://github.com/betaflight/betaflight/issues/6425#issuecomment-406921251. This is essentially the setup I am referring to in there.

wind0r, 2018-07-23:
This Feature Request / Issue is a duplicate of #4742

The RunCam Control Adapter Board also has TX/RX. Just having those PINs doesnt make it any smart. There is no backchannel since it only emulates the Joystick.

The Micro Eagle only has the Board build in to the casing.

Until there is some Camera with some real protocol there isnt any way to know when to disable and enable the BF OSD. Also it would be great to have some protocol instead of just up/down/left/right. Until then you need to add the OSD Enable Mode to some Aux Switch to enable disable it by hand

bforbort, 2018-07-24:
@wind0r According to the [wiki](https://github.com/betaflight/betaflight/wiki/FPV-Camera-Control-(Joystick-Emulation)), the throttle is kept at ~50% the entire time camera control is being used.  Could we disable the OSD when the first camera command is issued, and then enable the OSD again once the throttle is dropped to 0%?

[image]

DieHertz, 2018-07-24:
This may actually help, good thing I selected stick commands in such a way they use 50% throttle. I'll look into making a prototype

mikeller, 2018-07-24:
@wind0r:

> The Micro Eagle only has the Board build in to the casing.

As I said above, I suspect that the reason for this is that the chipsets that they are using for the cameras are security camera chipsets that just do not have any other control inputs than the 'keypad' interface. Having a custom chipset made is expensive - usually in the 100s of thousands, and I am not sure if the FPV camera market is big enough to be able to swallow this cost.

@bforbort: Good point, this could work!

closed 2018-08-30 by stale[bot]