betaflight/betaflight #6433: circual buffer for blackbox logging data

raimokr, 2018-07-23:
Some flight controllers only have some limited onbaord storage for blackbox data. For example the HolyBro Kakute V2 only has 16MB. With a bit higher blackbox sampling rates this lasts only for a very short amount of time (not even one pack).

I would like to have the possibility to use a circular buffer mechanism for the storage of bb data so that when the storage is full the oldest data permanently gets overwritten with the newest.

This way we can always have logging on and have access to the latest bb data.

mikeller, 2018-07-23:
Duplicate of #3343.

closed 2018-07-23 by mikeller

DieHertz, 2018-07-23:
If only it was practical we would have Implemented it long ago. What noticed how long blackbox erase takes? :)