betaflight/betaflight #6437: LIS3MDL support and list of supported mag/acc/gyro/etc

aster94, 2018-07-23:
hello,

I wasn't able to find an easy way to understand if the LIS3MDL was supported by BF (the answer is no). So i thought to open an issue with two request:

-support for this chip
-wiki page with a list of supported chip (mag/acc/gyro/etc)

mikeller, 2018-07-24:
@aster94: Do you have a board with this chip? Developers might be more inclined to help you adding a driver if they had hardware to develop against.

Have a look at this to get a list of the supported gyros:

https://github.com/betaflight/betaflight/blob/805b2f4b1ed3cf68ccbb0dd00d2ac8b1b999e198/src/main/sensors/gyro.c#L268-L408

aster94, 2018-07-24:
really no, i doesn't have any. i just saw them on ebay and checked some reviews and this mag seems interesting. On aliexpress it is sold at less then 2 dollars
thanks for the list, it is useful but not all the users here have basic programming knowledge, a simple wiki page which list all the suupported hardware from gyro.c, barometer.c, rangefinder.c, compass.c would be nice

mikeller, 2018-07-24:
@aster94: Feel free to add one - the wiki is free to edit for all GitHub users.

mikeller, 2018-07-24:
@McGiverGim might know this.

McGiverGim, 2018-07-24:
If I'm not wrong the 5983 is detected as 5883 and works. But I say this of memory. I will need to check the datasheet again.

aster94, 2018-07-24:
@McGiverGim ok i will wait your answer to add it
@mikeller i spend just a few minutes, it was a pleasure to help 😃

If you know other sensor which i didn t mention in the wiki please add them!

McGiverGim, 2018-07-24:
I can confirm that in theory (I don't have the hardware) the 5983 must work being detected as 5883. It will be great if some user could test it and verify, but by the moment, I think we can add it to the wiki.

bforbort, 2018-07-31:
I can probably help with this.  I already did the hard work when I [added LIS3MDL support for iNav](https://github.com/iNavFlight/inav/compare/master...bforbort:MAG_LIS3MDL).  Tested with the [mRo uGPS ublox SAM M8Q (GLONASS) + Compass (LIS3MDL)](https://store.mrobotics.io/product-p/mro-ugps-samm8q-01.htm) and I'm very happy with how well it's been working on my wing.

Shouldn't be too hard to port this code to Betaflight, I'll just have a couple of questions about how to set things up properly so that I don't cause issues with the Betaflight Configurator.

fiveangle, 2018-07-31:
@bforbort - what the ?  Guess https://github.com/iNavFlight/inav/issues/2607 can be closed ?

aster94, 2018-07-31:
@bforbort did you ever considered to do a comparison with the Hmc5983/hmc5883?

bforbort, 2018-08-01:
Just did a bench test on the compass driver and it worked with my [mRo uGPS ublox SAM M8Q (GLONASS) + Compass (LIS3MDL)](https://store.mrobotics.io/product-p/mro-ugps-samm8q-01.htm).

Does anyone else want to volunteer to test this before I do a PR?  You can either build [my tree](https://github.com/bforbort/betaflight/tree/LIS3MDL) by adding `USE_MAG_LIS3MDL` to your target.h file, and `compass_lis3mdl.c` to your target.mk file...

Or just let me know what target you use and I can post a .hex file here.

bforbort, 2018-08-02:
@mikeller There wouldn't happen to be a list of targets that have i2c solder pads, would there?  I don't want to add this driver to targets that would not benefit from it.

mikeller, 2018-08-02:
```
mikeller@pc:~/git/betaflight$ grep -l I2C src/main/target/*/target.h
src/main/target/AG3X/target.h
src/main/target/AIKONF4/target.h
src/main/target/AIR32/target.h
src/main/target/ALIENFLIGHTF1/target.h
src/main/target/ALIENFLIGHTF3/target.h
src/main/target/ALIENFLIGHTF4/target.h
src/main/target/ALIENFLIGHTNGF7/target.h
src/main/target/ALIENWHOOP/target.h
src/main/target/ANYFCF7/target.h
src/main/target/ANYFCM7/target.h
src/main/target/BEEROTORF4/target.h
src/main/target/BETAFLIGHTF3/target.h
src/main/target/BETAFLIGHTF4/target.h
src/main/target/BLUEJAYF4/target.h
src/main/target/CC3D/target.h
src/main/target/CHEBUZZF3/target.h
src/main/target/CJMCU/target.h
src/main/target/COLIBRI_RACE/target.h
src/main/target/COLIBRI/target.h
src/main/target/CRAZYFLIE2/target.h
src/main/target/DALRCF405/target.h
src/main/target/DALRCF722DUAL/target.h
src/main/target/EACHIF3/target.h
src/main/target/ELLE0/target.h
src/main/target/F4BY/target.h
src/main/target/FF_FORTINIF4/target.h
src/main/target/FOXEERF405/target.h
src/main/target/FRSKYF3/target.h
src/main/target/FURYF3/target.h
src/main/target/FURYF4/target.h
src/main/target/FURYF7/target.h
src/main/target/IRCFUSIONF3/target.h
src/main/target/ISHAPEDF3/target.h
src/main/target/KAKUTEF4/target.h
src/main/target/KAKUTEF7/target.h
src/main/target/KISSFC/target.h
src/main/target/KIWIF4/target.h
src/main/target/KROOZX/target.h
src/main/target/LUX_RACE/target.h
src/main/target/MATEKF405/target.h
src/main/target/MATEKF411/target.h
src/main/target/MATEKF722/target.h
src/main/target/MICROSCISKY/target.h
src/main/target/MIDELICF3/target.h
src/main/target/MOTOLABF4/target.h
src/main/target/MOTOLAB/target.h
src/main/target/MULTIFLITEPICO/target.h
src/main/target/NAZE/target.h
src/main/target/NERO/target.h
src/main/target/NUCLEOF446RE/target.h
src/main/target/NUCLEOF722/target.h
src/main/target/NUCLEOF7/target.h
src/main/target/OMNIBUSF4FW/target.h
src/main/target/OMNIBUSF4/target.h
src/main/target/OMNIBUSF7/target.h
src/main/target/OMNIBUS/target.h
src/main/target/OMNINXT/target.h
src/main/target/RCEXPLORERF3/target.h
src/main/target/REVOLT/target.h
src/main/target/REVONANO/target.h
src/main/target/REVO/target.h
src/main/target/RG_SSD_F3/target.h
src/main/target/SINGULARITY/target.h
src/main/target/SIRINFPV/target.h
src/main/target/SITL/target.h
src/main/target/SKYZONEF405/target.h
src/main/target/SPARKY2/target.h
src/main/target/SPARKY/target.h
src/main/target/SPEEDYBEEF4/target.h
src/main/target/SPEKTRUMF400/target.h
src/main/target/SPRACINGF3EVO/target.h
src/main/target/SPRACINGF3MINI/target.h
src/main/target/SPRACINGF3NEO/target.h
src/main/target/SPRACINGF3OSD/target.h
src/main/target/SPRACINGF3/target.h
src/main/target/SPRACINGF4EVO/target.h
src/main/target/SPRACINGF4NEO/target.h
src/main/target/SPRACINGF7DUAL/target.h
src/main/target/STM32F3DISCOVERY/target.h
src/main/target/STM32F7X2/target.h
src/main/target/VRRACE/target.h
src/main/target/X_RACERSPI/target.h
src/main/target/YUPIF7/target.h
```

tavdog, 2022-08-20:
Close bump

closed 2022-08-20 by haslinghuis