betaflight/betaflight #6441: RC filtering with CRSF

zytra, 2018-07-23:
Hi,
can someone post a BB explorer graph of what the RC commands should look like when RC filtering is enabled with CRSF protocol?
I was expecting smoother RC commands on the log, so I am wondering if I have it setup correctly.
thank you

etracer65, 2018-07-24:
With your TX and RX powered and bound, use the command `rc_smoothing_info` in the CLI and post the results here.  With Crossfire the automatic filter cutoffs should be 68Hz.  You can also provide a log if you want.

zytra, 2018-07-24:
let me run this command later tonight .

here's a screenshot of what the filtered RC command looks like:
[image]

zytra, 2018-07-25:
RC_SMOOTHING_INFO is picking up the right filters @ 67Hz.
not sure why I have these steps, or if they are normal for that matter

etracer65, 2018-07-25:
The log looks normal.  The auto settings are a compromise between too much (more delay) and too little (less smooth).

zytra, 2018-07-25:
thanks for the clarification!

closed 2018-07-25 by zytra