betaflight/betaflight #6448: Not accurate max values for the summary in the OSD

McGiverGim, 2018-07-25:
I don't know if this will have an easy solution...

The OSD stores and shows in the final summary some max values (altitude, amperage, etc.)

The problem is that the OSD rate is not too fast, so if we have some peaks, the peak is lost. For example, I can see in the Blackbox logs a peak of amperage of about 80A at the end of fast rolls,  but the OSD shows values usually between 50-60A.

I don't know if maybe is better to let store the max amperage to the current meter, that have all the information and let the OSD only read it, for example.

jflyper, 2018-07-25:
You want to see max values on OSD while flying?

If so, I would say a peak hold approach would be better.

mikeller, 2018-07-25:
@jflyper: But but but then the 80s will call, asking for their peak hold bars to be returned. :laughing:

etracer65, 2018-07-25:
It would make sense to do the same for min voltage as well.

McGiverGim, 2018-07-26:
True. But maybe this is not as fast in movements as the current meter, but yes, for coherence is better to change it for both.

etracer65, 2018-07-26:
@McGiverGim After looking at this for a bit I'm not sure there's anything we can or want to do to change this.

The primary "problem" you're seeing is that the blackbox log values for the voltage and current sensors are the unfiltered samples.  This is appropriate from a logging perspective but not useful from an end-user in-flight perspective.  In this case we want to use lowpass filtered values to smooth out noise and to slow down the rate of change caused by jitter or transient spikes.  Some large single sample spike in the blackbox is not representative of the real flight consumption.  It would be undesirable to have that large unfiltered spike affect the maximums reported on the stats page.  The current logic uses the filtered samples for both the in-flight and post-flight stats and I think this is more appropriate.  If there truly is a sustained current spike then it will get captured, but transients should be ignored.

Another problem that makes this difficult to change is that the current and voltage sensors are only sampled at 50hz.  On the other hand the OSD is running at (a scheduled) 60hz.  So unlike previously supposed there's not a case of the OSD "missing" samples.  It's all about seeing unfiltered data in the blackbox vs. filtered data in the OSD.

One thing that may help is the fix for the voltage/current sensor filter cutoffs in #6452.  The lowpass filters were being incorrectly set to a 20hz cutoff instead of 50hz.  So this would have impacted how quickly they responded and "smoothed" the peaks more than it should.

McGiverGim, 2018-07-26:
Ok, don't worry. I didn't know the filter thing. I close this because is not more an issue, it has some sense the results then.

Thanks for your time!!

closed 2018-07-26 by McGiverGim