betaflight/betaflight #6450: Incorrect filter settings for voltage and current sensor

rav-rav, 2018-07-25:
https://github.com/betaflight/betaflight/blame/master/src/main/sensors/current.c#L143
https://github.com/betaflight/betaflight/blame/master/src/main/sensors/voltage.c#L194

Both low pass filters are initialized with the refreshRate-parameter set to 50000. "refreshRate" is a little confusing, as it is actually "updateInterval".
The correct setting here would be 20000.
A fix could require additional changes to the filter frequency and should be tested. Unfortunately I don't have these sensors.

rav-rav, 2018-07-25:
I wasn't sure if it's a good idea when I can't test it. But I'll make one later.

ctzsnooze, 2018-07-25:
Rav-rav has dentified that the nomenclature we use for filter refreshRate is misleading.

refreshRate implies that we should be entering filter iteration frequency in Hz, but, if I understand it correctly, actually the value it wants is the interval in time between iterations.  

It would be really good to systematically change the name of refreshRate to something that would remove that confusion, and check that the values are appropriate, right across the code.

mikeller, 2018-07-26:
Leaving this open as a reminder to rename the refresh rate for 4.0.

mikeller, 2019-02-28:
Looking through the code, I think this has been addressed by now.
@rav-rav: Can this be closed?

rav-rav, 2019-03-01:
Yes, that is all good now.

closed 2019-03-01 by rav-rav