betaflight/betaflight #6451: Change adjustment rateprofile will not  be applied

achim1063, 2018-07-25:
Hi,
my "rate profile selection" adjustment can be applied to betaflight.
FC works fine with applied adjustments.
I changed the "using slot" parameter but fc still holds the first set paramters.
Reconnect shows the amendment but the FC behave with first set parameters.
Also a change of the rate profile, but doesn't solve this behavior.

Betaflight 3.3, configured 3 different rate Profiles for profile1.
FC Omnibus F4 Pro V3
Flysky FS-i6
Ragards,
Peter

etracer65, 2018-07-25:
After you land and disarm you need to save the settings with the stick command otherwise the changes will be lost when you disconnect the battery.

[image]

mikeller, 2018-07-26:
@achim1063: How do you try to switch rate profiles?

closed 2018-09-01 by stale[bot]