betaflight/betaflight #6453: X220 will only arm when connected to configurator

frozenskys, 2018-07-25:
**Describe the bug**
Wizard X220 will arm and spin the motors when connected to the configurator, but will go into an esc reboot loop if trying to arm (or sometimes when Tx switched on after powering quad) when not connected to configurator. 

**To Reproduce**
Connect quad to configurator, connect battery, on motor tab accept the safety message -> arm quad and move throttle, watch motors respond.

Remove battery, disconnect quad from configurator, connect battery -> arm quad and listen to esc reboot loop.

**Expected behavior**
When I arm the quad when not connected to the configurator then the motors arm and respond to Tx stick movements.

**Flight controller configuration**
```
# Betaflight / SPRACINGF3 (SRF3) 3.3.3 May 31 2018 / 13:34:34 (3565182d3) MSP API: 1.38
name X220

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature OSD
feature AIRMODE
feature ANTI_GRAVITY
serial 2 64 115200 57600 0 115200
aux 0 0 0 1925 2075 0
aux 1 1 1 1925 2075 0
rxfail 0 h
rxfail 1 h
rxfail 2 h
rxfail 3 h
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 5
set serialrx_provider = IBUS
set motor_pwm_protocol = DSHOT150
set failsafe_delay = 5
set failsafe_kill_switch = ON
set align_board_yaw = 90
set vbat_min_cell_voltage = 32
set osd_tim1 = 1280
set osd_vbat_pos = 2282
set osd_tim_1_pos = 2082
set osd_rtc_date_time_pos = 134
set osd_adjustment_range_pos = 68
set osd_core_temp_pos = 262
profile 0

set setpoint_relax_ratio = 40
set dterm_setpoint_weight = 108
rateprofile 3

```

**Setup / Versions**
 - Wizard X220 (SPF3)
 - Betaflight 3.3.3
 - Standard X220 Motors
 - Spedix ES25 ESCs
 - FS-IA6B Rx / FS-i6 Tx
 - Standard Wizard VTx

etracer65, 2018-07-25:
Sounds like a hardware problem - particularly the 5V power to the flight controller.  When connected via USB the flight controller (and possibly receiver) are powered via the USB port from your computer.  When only connected to the battery, the FC must get it's power from another source - usually a BEC on a PDB or possibly on-board depending on the flight controller and how you have it connected.  It sounds like that 5V supply is not stable.

frozenskys, 2018-07-25:
@etracer65 Thanks, I spent ages head scratching - I was convinced it was a hardware issue, but couldn't work out what would cause it. I'll do some voltage checks on the PDB when I get back to home tonight.

frozenskys, 2018-07-25:
@etracer65 spot on - the 5v BEC on the PDB is only giving 3.1v. Thanks for the help :)

closed 2018-07-25 by frozenskys