betaflight/betaflight #6454: OMNIBUSF4SD - F.Port on R9 Slim - No Telemetry but controls are ok

leonard4, 2018-07-25:
I have a Banggood Omnibus F4 V3 (with Sd Card) setup using a UBLOX spec GPS. I have the I2C remapped from UART3, I have the GPS on UART1, and I have F.port on the Sbus pin (Inverted UART6) with UART6 SerialRX turned on. I'm trying to use an R9 Slim that I have flashed to the latest F.port firmware. The only way I can get receiver input is with 'serialrx_inverted = on', and 'serialrx_halfdufplex = off', no other combination works, and after that I cannot get my transmitter to discover any sensors other than RxBT and RSSI for telemetry. 

Is this an issue with the build for this board or do I need to do some weird remap to enable some softserial somewhere? Like Motor5 remapped to a Softserial1 TX (or rx?) with halfduplex on and inversion off? I  haven't had any luck finding any info of anyone else successfully using an R9 receiver with F.port on BF yet.

`# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.4.0 Jul  8 2018 / 04:24:36 (cce8a59be) MSP API: 1.39

board_name OMNIBUSF4SD
manufacturer_id 

# name
name Nighthawk

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_RX 12 NONE
resource INVERTER 1 NONE
resource INVERTER 2 NONE
resource INVERTER 3 C09
[... 712 lines cut ...]
set horizon_tilt_effect = 75
set horizon_tilt_expert_mode = OFF
set abs_control_gain = 0
set abs_control_limit = 90
set abs_control_error_limit = 20

# rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = OFF
set throttle_limit_percent = 100`

bokie1, 2018-07-25:
Use the uninverted s.port pad on your r9 slim and connect it to uart6 tx.

leonard4, 2018-07-25:
There is no uninverted pad on this R9 Slim, the one pad that I found online does not exist on my version. I wish it did. I just pulled the Omnibus out and put in a Bardwell F4, got everything working so far, only thing I lost was a barometer which I don't need with GPS altitude. Going to make this as closed since I've moved on from this board.

bokie1, 2018-07-25:
I have one of the first r9 slim and it has an uninverted pad.

leonard4, 2018-07-27:
Any chance you can show a picture to where the pad is?

jflyper, 2018-07-29:
Closed as this is not a BF software issue.

closed 2018-07-29 by jflyper

rmawatson, 2019-06-01:
@leonard4  Did you ever get this working?