betaflight/betaflight #6460: Adding Magnetometer Support to CLRACINGF4 target (Bardwell F4), missing a small step...

leonard4, 2018-07-26:
Had a working OMNIBUSF4SD with Magnetometer and GPS in BF 3.4, it appears that the MAG/I2C code is not present in the CLRACINGF4 target and I've been adding it in by hand but I think I missed something since its not working. 

MAG button at the top does not light up, and the checkbox for Magnetometer in the sensors page is greyed out.

`# set mag
align_mag = DEFAULT
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

mag_bustype = I2C
Allowed values: NONE, I2C, SPI, SLAVE

mag_i2c_device = 2
Allowed range: 0 - 3

mag_i2c_address = 0
Allowed range: 0 - 119

mag_spi_device = 0
Allowed range: 0 - 3

mag_hardware = AUTO
Allowed values: AUTO, NONE, HMC5883, AK8975, AK8963, QMC5883

mag_declination = 0
Allowed range: -18000 - 18000

mag_calibration = 0,0,0
Array length: 3`

This is the fork I have: https://github.com/leonard4/betaflight/tree/master/src/main/target/CLRACINGF4

I've added the two compass C files to the target.mk file, and the USE_I2C and USE_MAG lines to the target.h file. I think I'm missing something else to get it to fully work and was hoping someone could point me in the right direction!

jflyper, 2018-07-26:
Well, ... these two lines should be

https://github.com/leonard4/betaflight/blob/23214ca159918e4a2df6f5caba7e2cd12ace9578/src/main/target/CLRACINGF4/target.h#L141-L142
```
#define I2C2_SCL                C06 // shared with UART6TX
#define I2C2_SDA                C07 // shared with UART6RX
```

`C06` -> `PC6`
`C07` -> `PC7`

but the fundamental problem is that these pins are not I2C capable.

Refer to Table.9 of DM00037051.pdf (search on the net) for pins that are I2C capable.

For further assistance, please use other places on the net, as this is not an issue with the Betaflight software.

leonard4, 2018-07-27:
Thanks for the doc, will check that over. Wasn't aware that there were certain I2C capable pins, learning something knew every day! :)

closed 2018-07-27 by leonard4

VladTrach, 2020-06-24:
Has anyone managed to get an i2c port for Bardwell f4 v2?

haslinghuis, 2020-06-25:
Ask @joshuabardwell 😉 j/k did not find i2c in his manual