betaflight/betaflight #6462: DShot Arming + Flip Causes Unpredictable Arming

bforbort, 2018-07-27:
To limit the number of channels I use, I've programmed my radio's momentary "trainer" switch as a panic switch.  It activates Angle mode, Flip Over After Crash, and Beeper.  In flight, it will level the quad; when disarmed, it will activate flip and beeper.

**Describe the bug**
With DShot beacon enabled, when I try to activate Flip Over After Crash, if I do not hold the switch active for at least a couple of seconds, there will be a short delay, then the quad will activate in normal flight mode.

This can lead to unpredictable arming behavior when upside-down or in trees.

**To Reproduce**
Enable DShot Beacon.  Enable modes Beeper and Flip Over After Crash on the same switch. Attempt to arm in Flip mode, but set the Flip switch to off immediately after arming. 

**Expected behavior**
Flip Over After Crash should always activate when it's switch is activated while arming, regardless of Beacon being active or not.

**Additional context**
Here is a [video of the current and expected behavior](https://youtu.be/Wrfl38UM6cQ).

**Flight controller configuration**
Note: Trainer switch is aux 3
``` 
# version
# Betaflight / AG3XF4 (AGX4) 3.5.0 Jul 25 2018 / 10:12:09 (74d883bfa) MSP API: 1.40

board_name AG3XF4
manufacturer_id 

# name
name K1QV CG

# resources

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL
feature TELEMETRY
feature AIRMODE

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 1 3 1700 2100 0 0
aux 3 2 1 1300 1700 0 0
aux 4 13 3 1700 2100 0 0
aux 5 26 0 1700 2100 0 0
aux 6 35 3 1700 2100 0 0
aux 7 39 4 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass2_hz = 160
set acc_trim_roll = 2
set acc_calibration = 14,-33,342
set mag_hardware = NONE
set rssi_channel = 12
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 20
set serialrx_provider = CRSF
set dshot_idle_value = 500
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 32000
set align_board_yaw = 270
set vbat_max_cell_voltage = 44
set vbat_min_cell_voltage = 34
set vbat_scale = 108
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1550
set osd_tim2 = 1537
set osd_rssi_pos = 2115
set osd_tim_2_pos = 2134
set osd_anti_gravity_pos = 2337
set osd_throttle_pos = 2241
set osd_ah_pos = 173
set osd_current_pos = 2272
set osd_mah_drawn_pos = 2306
set osd_altitude_pos = 2294
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2123
set osd_disarmed_pos = 2218
set osd_nvario_pos = 2264
set osd_esc_tmp_pos = 2160
set osd_esc_rpm_pos = 2167
set osd_rtc_date_time_pos = 2436
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set vtx_pit_mode_freq = 5917
set vcd_video_system = NTSC

# profile
profile 0

set crash_recovery = ON
set iterm_relax = RP
set yaw_lowpass_hz = 100
set p_pitch = 44
set d_pitch = 14
set p_roll = 42
set d_roll = 13
set p_yaw = 60

# rateprofile
rateprofile 0
```

bforbort, 2018-07-27:
Is it possible this issue was caused by this 1e6e9e719f94403a0f4cafaa4198b91b49e91614 commit?

mikeller, 2018-07-27:
Yes, it looks like this is an unintended side effect of 1e6e9e7 . And I agree that the more intuitive behaviour would be for modes that latch on arming to latch on the state that is active when the arming switch is activated, not when the actual arming happens.

We will need to look into a fix, not sure how much effort this will be.

etracer65, 2018-08-06:
@bforbort I have a fix for this issue in https://github.com/betaflight/betaflight/pull/6525

Can you try this version to confirm it fixes the problem?

[betaflight_3.5.0_AG3XF4.hex.zip](https://github.com/betaflight/betaflight/files/2263907/betaflight_3.5.0_AG3XF4.hex.zip)

As is always a good practice, **Perform initial tests without props.**

bforbort, 2018-08-06:
@etracer65 That did correct the issue.  Quad arms as expected regardless of how long I hold the mode switch.

Thanks much!

PR #6525 (2018-08-08): Capture and use crash flip if set initially when arming is delayed due to beacon
Fixes #6462 

Previously if arming was delayed due to DSHOT beacon then the arming type (normal or crash-flip) would be captured **after** the delay. This change captures the state on entering the delay which more accurately represents the pilot's intent.

For example, the pilot may have crash-flip on a momentary switch that's initially set when the arming switch is enabled. Previously if the pilot didn't hold the momentary switch through the entire delay the quad would arm in normal mode. This change uses the sta [... cut]
closed 2018-08-06 by bforbort