betaflight/betaflight #6470: 3.5RC1: Servo stops responding on tricopter just after gyro calibration

brycedjohnson, 2018-07-28:
Wanted to try out the new yaw FF feature with an old tricopter to see if it makes it more stable.  Using the newest configurator from today. 

with `tri_unarmed_servo = ON` I can move the servo around before the gyro calibrates.  After I get the beeps for the gyro calibration I can't move it anymore.

https://youtu.be/0S5anshHjMU

Also at that moment the image of the model moving around in BF setup (when the acc is on) jumps back to heading 0 pitch 0 roll 0.  On the Sensor tab I get see each gyro axis go to 0.  Acc keeps working.  Could be odd bad hardware but I did just have it flying on 3.4RC5 a couple weeks ago. And just tested with 3.4 official on the bench with no issues.

Can't arm (unless I turn acc off) due to `angle` even though I have small angle set to 180.

After arming none of the pitch, roll, yaw do anything.  The switch to arm/disarm still works.

# diff all
# Betaflight / SPRACINGF3 (SRF3) 3.5.0 Jul 25 2018 / 10:48:04 (74d883bfa) MSP API: 1.40

mcu_id 002300275233570420333531
defaults nosave
resource MOTOR 1 A11
resource MOTOR 2 A06
resource MOTOR 3 A07
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource SERVO 1 A02
mixer TRI

feature -RX_PPM
feature RX_SERIAL
serial 1 32 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
aux 0 0 0 1200 1650 0 0
aux 1 13 1 1700 2100 0 0
aux 2 26 2 1300 2100 0 0
aux 3 35 1 1300 1700 0 0
aux 4 36 3 1300 2100 0 0
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 8000
set align_board_yaw = 90
set small_angle = 180
set pid_process_denom = 1
set report_cell_voltage = ON
profile 0

set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 8
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100
profile 1

profile 2

profile 0
rateprofile 0

set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 76
rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

rateprofile 0
save

DieHertz, 2018-07-28:
Disable dynamic filter and report back

brycedjohnson, 2018-07-28:
@DieHertz Yup that was it!  Disabled notch and can could fly.

DieHertz, 2018-07-28:
Then we know about this bug an working on resolving it. After increasing dynamic filter sampling to 1333 Hz it stopped working for 2KHz and below. Before it was not working for 1KHz and below.

What looptime you were running at? Your log suggested equal PID and gyro frequencies, but which exactly? Was it 2/2?

brycedjohnson, 2018-07-28:
@DieHertz, Ahh that makes senses, I am at 2k/2k, damn nyquist.  I'm pulling some some logs right now, but I can give it a try at 4k/4k.  Must have been running a custom build with some debug info awhile back and just had so I had to bump it down to 2k.

DieHertz, 2018-07-28:
2.67KHz and above will work okay

brycedjohnson, 2018-07-28:
Just tried 2.67k, that worked as well.

mikeller, 2018-07-28:
@brycedjohnson: Can you test how `yaw_d` works on tricopter? That was added mainly with the tricopter use case in mind.

brycedjohnson, 2018-07-28:
@mikeller That and the FF was why I wanted to try it out.  I've never got the rcexplorer v3 tuned very well, but thought I'd give it another go.  Getting closer with the logs and the pid analyzer.

Couple logs if you are interested:
[tricopter.zip](https://github.com/betaflight/betaflight/files/2238326/tricopter.zip)

mikeller, 2018-07-29:
@brycedjohnson: Definitely interested to hear the results. I always found that getting a reliable tune that did not make the craft prone to tail wobble was touch and go. I hope that re-adding yaw D and smart FF will make this better. I'll have to upgrade the F1 flight controller in my tricopter before I can try...

etracer65, 2018-07-29:
@brycedjohnson Try #6485 when you have a chance. Should fix the problem with dynamic filter at 2KHz.  But we still recommend running higher than 2KHz as it make the dynamic filter much more effective.

PR #6485 (2018-08-02): Fix dynamic filter for gyro loop < 4K
Fixes #6470 

For gyro loops from 2K to < 4K scale the FFT_SAMPLING_RATE_HZ to ensure we get 3 samples.

Also constrain the `centerFreq` result from the lowpass to ensure it doesn't overshoot beyond the min/max notch center cutoff.
closed 2018-08-02 by mikeller