betaflight/betaflight #6472: MSP Subscription

awolf78, 2018-07-29:
I currently have to hammer through 9-10 MSP requests every loop to retrieve all the data I want for ImpulseOSD. Instead of sending MSP requests and waiting for their response I would like to add an MSP subscription where I tell betaflight which MSPs I want and it will start sending them in a decent frequency (100 ms or so). They would all come in one byte stream with the usual MSP protocol info (MSP type and length), header would only be in the front. XOR CRC can be for each MSP or one at the very end.

Before I dive into the code and start hacking this myself and start making a PR I wanted to discuss this first:
Do you have something similar like that already planned?
If not: Are you OK if I do this and do you have any idea/requirements how you would like this to look like?

awolf78, 2018-07-29:
Not trying to beat a dead horse, but my proposal would not create any compatibility issues, so here it is in more detail:

I would add one more MSP IN (CONFIGURE_SUBSCRIPTION):

- Byte 1+2 (uint16_t): Desired frequency from 10ms-10000ms; this is not guaranteed, but true in average
- Byte 1..n: The desired MSP codes (max: 20)
- Sending the same command with a length of 0 will cancel all active subscriptions

The FC would then fill a buffer with one MSP every FC loop according to the configuration above. Once all the MSPs are inside the buffer (and the frequency criteria is met) it would give the DMA a nudge (if it is not busy) and send them off. The receiving side must take care that it can receive the MSPs in the frequency they are sent to not receive incomplete data.

Maybe I am missing something, but I do not see any compatibility issues with this approach. Have not looked in detail how the MSP requested are queued, but if there is a queue, then the subscriptions would take a lower priority to the other incoming MSP requests.

mikeller, 2018-07-30:
So is your proposal to only send these unrequested replies only on the port that CONFIGURE_SUBSCRIPTION was received on? If so, how would you handle concurrent MSP requests on other ports?

> Have not looked in detail how the MSP requested are queued,

They are not, currently, the only queueing is the RX queue of the UART. That is part of why this will likely cause compatibility issues.

awolf78, 2018-07-30:
It does not make sense to broadcast a subscription to a port it was not requested from. 

Looking a bit deeper in the code (msp_serial.c), it could be a possibility to add the subscriptions as a "Pending Request":
```C
static void mspProcessPendingRequest(mspPort_t * mspPort)
{
    // If no request is pending or 100ms guard time has not elapsed - do nothing
    if ((mspPort->pendingRequest == MSP_PENDING_NONE) || (millis() - mspPort->lastActivityMs < 100)) {
        return;
    }

    switch(mspPort->pendingRequest) {
        case MSP_PENDING_BOOTLOADER:
            systemResetToBootloader();
            break;

#ifdef USE_CLI
        case MSP_PENDING_CLI:
            cliEnter(mspPort->port);
            break;
#endif
        case MSP_PENDING_SUBSCRIPTION:
             processSubscriptions(mspPort->port);
        default:
            break;
    }
}
```

mspProcessPendingRequest is called when no byte is in the RX queue for that particular msp Port:
```C
void mspSerialProcess(mspEvaluateNonMspData_e evaluateNonMspData, mspProcessCommandFnPtr mspProcessCommandFn, mspProcessReplyFnPtr mspProcessReplyFn)
{
    for (uint8_t portIndex = 0; portIndex < MAX_MSP_PORT_COUNT; portIndex++) {
        mspPort_t * const mspPort = &mspPorts[portIndex];
        if (!mspPort->port) {
            continue;
        }

        mspPostProcessFnPtr mspPostProcessFn = NULL;

        if (serialRxBytesWaiting(mspPort->port)) {
            // There are bytes incoming - abort pending request
            mspPort->lastActivityMs = millis();
            mspPort->pendingRequest = MSP_PENDING_NONE;

            while (serialRxBytesWaiting(mspPort->port)) {
                const uint8_t c = serialRead(mspPort->port);
                const bool consumed = mspSerialProcessReceivedData(mspPort, c);

                if (!consumed && evaluateNonMspData == MSP_EVALUATE_NON_MSP_DATA) {
                    mspEvaluateNonMspData(mspPort, c);
                }

                if (mspPort->c_state == MSP_COMMAND_RECEIVED) {
                    if (mspPort->packetType == MSP_PACKET_COMMAND) {
                        mspPostProcessFn = mspSerialProcessReceivedCommand(mspPort, mspProcessCommandFn);
                    } else if (mspPort->packetType == MSP_PACKET_REPLY) {
                        mspSerialProcessReceivedReply(mspPort, mspProcessReplyFn);
                    }

                    mspPort->c_state = MSP_IDLE;
                    break; // process one command at a time so as not to block.
                }
            }

            if (mspPostProcessFn) {
                waitForSerialPortToFinishTransmitting(mspPort->port);
                mspPostProcessFn(mspPort->port);
            }
        }
        else {
            mspProcessPendingRequest(mspPort);
        }
    }
}
```

Note that:

- Each port is called separately
- Any receiving byte has priority over mspProcessPendingRequest

The idea would be that the subscriptions have a "UDP style" behavior: It is not guaranteed that clients will receive the entire message every single time - if they start sending MSP requests on the same port to the FC, the subscription will be interrupted and the current MSP request gains priority. Subscriptions will only be fired when the line is free.

hydra, 2018-07-30:
yeah, the idea is very similar to the MSP stream code i wrote a while ago.

i certainly have no problem with the idea.

i think i had msp-stream-start and msp stream stop commands. start took took two params, freq and msp id.  stop took optional msp id, stop without any id to stop all streamed commands.

MSP telemetry basically already did the sending stuff so i felt it was just a rework of that.

awolf78, 2018-07-30:
@hydra Great, does the MSP stream support multiple MSPs?
Does your code still exists? I did some digging, but could not find anything in CF/BF.

hydra, 2018-07-30:
Yes, I found it here:

https://github.com/cleanflight/cleanflight/pull/2763

hydra, 2018-07-30:
The code linked above, was working just fine in practice.  Not sure if any of it is usable now, but at least you can see the changes that had to be made.

Your usage scenario is not quite the same, in my usage scenario the OSD was the SERVER of information (current sensor) and the FC was the CLIENT.  In your usage scenario the FC is the SERVER and your OSD is the CLIENT.  In practice it's identical since the CLIENT and SERVER code were both implemented in CF/BF with my PR.  You would just need to replicate the CLIENT code in your OSD and the SERVER code in CF/BF with your new PR.  You don't need to implement the CLIENT in CF/BF at this time, as that was only used by the OSD SLAVE which is a special build of CF/BF without all the flight code.

awolf78, 2018-07-30:
@hydra Thank you very much for the reference to your PR/discussion, that cleared things up quite a bit for me.

@martinbudden and @digitalentity and @mikeller 

Let me modify my approach:
Instead of doing "unsolicited updates" I have no problem changing this to a polling approach. My main motivation is to reduce the amount of ping-pong requests from 9-10 or more MSPs down to one. If I have to request that every time, that is fine with me:

- Client (in this case ImpulseOSD) sends a request via a new MSP (name to be defined) which includes all the MSPs requested
- Server (in this case FC) attaches them all together in one byte stream and sends them to the client
- Client will only receive data from Server as long as it keeps polling the new MSP containing the MSP codes requested. ONCE.

This approach should significantly increase the refresh rate of the (telemetry) data. There would be also minimized time gaps between the individual MSPs, which can cause adverse effects. I have e.g. throttle based amperage statistics and the data for retrieving the throttle is in a different MSP as the data for the amperage.

hydra, 2018-07-30:
@awolf78 I'm concerned that there would not be enough memory for the server to send all the data and that the UART TX buffer would overrun.  Unless of course the TX buffer isn't an issue these days?  There's also the amount of time it would take the FC to collect all the data required, if done in a blocking way this could create scheduling issues if it takes too long.

hydra, 2018-07-31:
@awolf78 yes it would need to run on F3.  All the SPRACINGF3OSD + SPRACINGF3 ACRO/DELUXE/SPRACINGF3EVO/SPRACINGF4EVO targets (which all stack) would benefit from this change.

awolf78, 2018-07-31:
Just compiled 3.4 SPRACINGF3 target:
```
Linking SPRACINGF3
Memory region         Used Size  Region Size  %age Used
           FLASH:      252608 B       252 KB     97.89%
    FLASH_CONFIG:          0 GB         4 KB      0.00%
             RAM:       37508 B        40 KB     91.57%

Not much room left, especially RAM wise. Would need to work with the buffers that are available and take care not to overflow the buffer if too many MSPs are requested. These would need to get omitted since there is barely any room for stack left. Cannot really afford any more buffers on an F3.

ledvinap, 2018-07-31:
In my experience it is more maintainable to use polling approach (keeping protocol simple). 
100ms is long time, you can prepare list of requested packets on OSD and send it to FC periodically ...

awolf78, 2018-07-31:
Polling approach it is, I am glad we went already past that.
I am actually looking to pull data the other way: from FC to OSD.

awolf78, 2018-08-01:
Is it OK to change
```
void sbufWriteU8(sbuf_t *dst, uint8_t val)
{
    *dst->ptr++ = val;
}
```

to

```
void sbufWriteU8(sbuf_t *dst, uint8_t val)
{
    if (dst->ptr != dst->end) *dst->ptr++ = val;
}
```

Then there will be never a buffer overflow. The buffer may contain incomplete data, but that is much better than overflowing the buffer.

ledvinap, 2018-08-01:
@awolf78 : `sbuf` is used in a lot of places, performance penalty may be non-negligible.

It may be better to check for available space in explicitly, possibly only when starting some chunk ...

awolf78, 2018-08-01:
@ledvinap That is what I thought. I came up with a different idea in the meantime that avoids this. I am checking how many MSP requests would fit in the dst buffer by writing them in there one by one. I am building the sum of the bytes that are written and reset the pointer of dst to the beginning for every MSP. Then I start sending all the MSPs up to the MSP that will fit in the dst buffer.
Need to do some more testing, but if I don't hear any complaints about my approach I will start a PR soon.

awolf78, 2018-08-02:
Doesn't look good for the F3 in my initial testing. Since some of the MSP have dynamic sizes I cannot rely on a cache to prevent buffer overflows. I have to check the size of each MSP by executing it first and then write them in the buffer again afterwards - up to the MSP that fits in the buffer.
Will do more testing on the F4 in the next few days.

mikeller, 2018-08-03:
Yeah, F3 is a constant pain point these days: #6444, #6449, #6497, #6501.

awolf78, 2018-08-03:
Shame on me, good ol' F3 can do it. Issue was in front of the computer.

awolf78, 2018-08-06:
Implemented with https://github.com/betaflight/betaflight/pull/6509

closed 2018-08-06 by awolf78