betaflight/betaflight #6474: 3.5 Rc1 set throttle_boost stays at 0 no matter what you input.

chime13, 2018-07-29:
# version
# Betaflight / SPRACINGF3EVO (SPEV) 3.5.0 Jul 25 2018 / 10:48:16 (74d883bfa) MSP API: 1.40

In CLI, no matter what profile or setting I use for Throttle_boost, I get the following:

# set throttle_boost
throttle_boost = 0      // I tried to use a value of 20
profile 2
Allowed range: 0 - 100
Default value: 5

throttle_boost_cutoff = 15
profile 2
Allowed range: 5 - 50

Here's a Diff - As you can see, Throttle_boost comes up as 0
_______________
# diff
# Betaflight / SPRACINGF3EVO (SPEV) 3.5.0 Jul 25 2018 / 10:48:16 (74d883bfa) MSP API: 1.40

resource PWM 1 D07
resource PWM 2 C08
resource PWM 3 K05
resource PWM 4 D04
resource PWM 5 B02
resource PWM 7 B01
resource PWM 8 A02

feature -RX_PPM
feature -TELEMETRY
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature MOTOR_STOP
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
map TAER1234
serial 2 64 115200 57600 0 115200
aux 0 28 1 1300 2100 0 0
aux 1 30 2 1700 2100 0 0
adjrange 0 0 0 900 2100 12 0 0 0
rxrange 0 1090 1907
rxrange 1 1090 1907
rxrange 2 1090 1907
rxrange 3 1090 1907
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_src_frame_errors = ON
set rssi_scale = 41
set rc_interp = OFF
set rc_interp_ch = RP
set fpv_mix_degrees = 15
set serialrx_provider = SPEK2048
set spektrum_sat_bind_autoreset = OFF
set min_throttle = 1010
set motor_pwm_rate = 32000
set yaw_motors_reversed = ON
set pid_process_denom = 2
set rate_6pos_switch = ON
profile 2

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 20000
set feedforward_transition = 50
set smart_feedforward = ON
set throttle_boost = 0
set p_pitch = 157
set d_pitch = 67
set f_pitch = 0
set p_roll = 155
set d_roll = 65
set f_roll = 0
set p_yaw = 180
set f_yaw = 0
rateprofile 2

set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150
set roll_expo = 40
set pitch_expo = 40

mikeller, 2018-07-29:
@chime13:

```
set rc_interp_ch = RP
```

Throttle boost needs interpolation to be on for throttle, use `set rc_interp_ch = RPT` to get it going.

chime13, 2018-07-29:
Thanks @mikeller 

It worked only after setting rc_interp = AUTO
in addition to setting rc_interp_ch = RPT

closed 2018-07-29 by chime13