betaflight/betaflight #6476: Weighted averaging gyros data on FC which has different gyro and cs

DusKing1, 2018-07-29:
I noticed that some FC-board productor are testing some dual gyro F7 FC（like the DALRC inside-developing F7dual v2 board）. These FC has two different gyros :one is like ICM20602 and the other is MPU6000. And these FC has cs.
But the embarrassing thing is that there FC can only use one gyro data at the same time.

Could you please develop a weighted average gyro data algorithm to melt the gyro data and to improve these FC‘s performance？

jflyper, 2018-07-29:
IMO, combining data from MPU6000 and ICM20602 does not deliver significant improvement over MPU6000 alone.

- MPU6000 can not deliver data at 32KHz.
- MPU6000 can deliver very good data at 8KHz.
- So does ICM20602.

DusKing1, 2018-07-29:
@jflyper I kown the data delivered by MPU6000 is VERY good, but IMO, the higher sampling rate the better signal fidelity in theory, right?

To be honest, I always do my flying with MPU6000-FC and I'm pretty satisfied with its performance. But once I saw a video which shows the different scenario between gyro@32K and gyro@8k at one hardware platform (just switching gyro data sampling rate of icm20602),I had to admire that gyro@32K(or I say just higher sampling rate than 8k) really bring some improvements.(Or my feeling is wrong? Plz help me correct my opinion.)

BTW, if the performance of ICM20602 @16k or@32k is not better than @8k, why do spracingf7dual-like fc with dual gyro exist ? (Just a little bit confused.)

jflyper, 2018-07-29:
@DusKing1 
At time 0, MPU6000 delivers M1 and ICM delivers data S1.
At time 31.25usec, ICM delivers new data S2. MPU6000 may be read, but it will probably deliver M1 again.
What would be a combined data at 32.25usec?

DusKing1, 2018-07-29:
Well, I think I kown where the unreasonable part is. Sorry for unreasonable feature request

closed 2018-07-29 by DusKing1