betaflight/betaflight #6477: 3.5.0 RC1 Crash Protection returns attitude control very slowly.

babailey, 2018-07-29:
**Describe the bug**
after hitting a gate, the quad held attitude for several seconds before very gradually giving control back to RC inpus.

**To Reproduce**
Set Crash_Recovery = ON, hit something.

**Expected behavior**
Recover and give control back. 

**Flight controller configuration**
Create a `diff` and post it here in a code block. 
(note: I disabled crash recovery in this DIFF, everything else is as it was when the issue occurred)
``` 
# diff

# version
# Betaflight / PYRODRONEF4 (PYRO) 3.5.0 Jul 25 2018 / 10:42:45 (74d883bfa) MSP API: 1.40

board_name PYRODRONEF4
manufacturer_id 

# name
name Frosty+

# resources

# mixer
mixer QUADP

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE

# beeper

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led
led 0 5,8::C:8
led 1 6,8::C:8
led 2 7,8::C:8
led 3 8,8::C:8
led 4 9,8::C:8
led 5 10,8::C:8
led 6 11,8::C:8

# color

# mode_color

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 3 1825 2100 0 0
aux 2 35 1 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set dyn_notch_quality = 15
set dyn_notch_width_percent = 30
set min_check = 1005
set max_check = 1995
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set dshot_idle_value = 220
set motor_pwm_protocol = DSHOT600
set align_board_roll = 180
set align_board_yaw = 315
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2072
set osd_craft_name_pos = 2432
set osd_warnings_pos = 2218
set osd_avg_cell_voltage_pos = 2104
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set vtx_channel = 4
set vtx_freq = 5800
set vcd_video_system = NTSC

# profile
profile 0

set dterm_notch_cutoff = 0
set iterm_relax = RP
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# rateprofile
rateprofile 0

set roll_rc_rate = 88
set pitch_rc_rate = 88
set yaw_rc_rate = 88
set roll_expo = 5
set pitch_expo = 5
set yaw_expo = 5
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65  ``` 

**Setup / Versions**
PyroF4OSD
XM+ / SBUS
Taranis X9D+

**Additional context**
This happened multiple times, did not happen after disabling crash recovery.

mikeller, 2018-07-30:
@babailey: You don't even have crash recover enabled, so this must be something else.

babailey, 2018-07-30:
[image]

mikeller, 2018-07-30:
Can you please post a `diff` that works to replicate the problem? Diagnosing stuff based on guestimates is a waste of time.

Footr, 2018-08-11:
I'm having the same issue.  When crash recovery activates it feels almost like a failsafe.  You used to be able to grab control from it and it would help you... it now feels like you are locked out of the quad.  

`# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.0 Aug  2 2018 / 01:53:03 (d0a9a0fd9) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 
mcu_id 003000403037471132363731
signature 

# reset configuration to default settings
defaults nosave

# name
name Footr3

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B08
resource LED 1 NONE

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 28 3 1775 2100 0 0
aux 3 35 3 1400 1650 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 120
set min_check = 1020
set max_check = 1990
set rc_smoothing_type = FILTER
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1050
set vbat_max_cell_voltage = 45
set vbat_warning_cell_voltage = 32
set ibata_scale = 254
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 1200
set osd_tim1 = 529
set osd_vbat_pos = 44
set osd_rssi_pos = 40
set osd_tim_1_pos = 2442
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_pos = 200
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2424
set osd_craft_name_pos = 2401
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 9
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set vtx_band = 5
set vtx_channel = 8
set vtx_freq = 5917
set camera_control_internal_resistance = 480

# profile
profile 0

set dterm_lowpass_hz = 80
set dterm_lowpass2_hz = 160
set dterm_notch_cutoff = 0
set feedforward_transition = 10
set crash_gthreshold = 550
set crash_recovery = ON
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 51
set i_pitch = 62
set d_pitch = 22
set f_pitch = 100
set p_roll = 50
set i_roll = 58
set d_roll = 20
set f_roll = 100
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 105
set pitch_rc_rate = 90
set yaw_rc_rate = 65
set roll_expo = 1
set pitch_expo = 2
set roll_srate = 58
set pitch_srate = 38
set yaw_srate = 43
set tpa_breakpoint = 1550

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

Footr, 2018-08-11:
Same thing is happening to me.  When it activates it feels like a failsafe.  You have no input. No control.  Just feels like the quad is totally doing its thing.  Been using crash recovery since the beginning, never felt like this before.    

[BTFL_cli_Footr3_20180811_133952.txt](https://github.com/betaflight/betaflight/files/2280424/BTFL_cli_Footr3_20180811_133952.txt)

Footr, 2018-08-11:
I just went out a flew a few packs.  I clipped a tree, crash recovery activated and leveled, and never gave me control back on the pitch and roll axis.  I was still able to yaw so I was able to bring the quad in.  Flew for about 5-10 seconds like this.  Auto level was locked and I could only yaw.  

```diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.0 Aug  2 2018 / 01:53:03 (d0a9a0fd9) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 
mcu_id 003d00343037470c32363731
signature 

# reset configuration to default settings
defaults nosave

# name
name Footr2

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B08
resource LED 1 NONE

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 28 3 1775 2100 0 0
aux 3 35 3 1400 1650 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 120
set dyn_notch_quality = 15
set dyn_notch_width_percent = 30
set min_check = 1020
set max_check = 1990
set rc_smoothing_type = FILTER
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1050
set vbat_max_cell_voltage = 45
set vbat_warning_cell_voltage = 32
set ibata_scale = 254
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 1200
set osd_tim1 = 529
set osd_vbat_pos = 44
set osd_rssi_pos = 40
set osd_tim_1_pos = 2442
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_pos = 200
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2424
set osd_craft_name_pos = 2401
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 9
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5917
set camera_control_internal_resistance = 480

# profile
profile 0

set dterm_lowpass_hz = 80
set dterm_lowpass2_hz = 160
set dterm_notch_cutoff = 0
set crash_gthreshold = 550
set crash_time = 250
set crash_recovery = ON
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 51
set i_pitch = 62
set d_pitch = 22
set f_pitch = 100
set p_roll = 50
set i_roll = 58
set d_roll = 20
set f_roll = 100
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 105
set pitch_rc_rate = 90
set yaw_rc_rate = 65
set roll_expo = 1
set pitch_expo = 2
set roll_srate = 58
set pitch_srate = 38
set yaw_srate = 43
set tpa_breakpoint = 1550

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

AliB10, 2018-08-14:
i get the same thing.
Clip a gate, scuff the floor crash recovery will enable and then my pitch / roll is disable and i only have yaw and throttle. This can last up to 20s of hovering before coming back. Or land, disarm, re arm and you're good to go again.

[code] version
# Betaflight / OMNIBUSF4SD (OBSD) 3.5.0 Aug  2 2018 / 02:14:24 (d0a9a0fd9) MSP API: 1.40

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources
resource MOTOR 2 A03
resource MOTOR 3 B01

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -OSD
feature RX_SERIAL
feature TELEMETRY
feature AIRMODE

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 2100 0 0
aux 1 35 2 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1000

# master
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1950
set rc_smoothing_type = FILTER
set serialrx_provider = CRSF
set airmode_start_throttle_percent = 20
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set feedforward_transition = 30
set crash_recovery = ON
set iterm_relax = RP
set i_pitch = 90
set d_pitch = 25
set f_pitch = 80
set i_roll = 85
set f_roll = 80
set i_yaw = 85

# rateprofile
rateprofile 0

set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 90
set roll_srate = 35
set pitch_srate = 35
set yaw_srate = 50 [/code]

si618, 2018-08-30:
Hopefully this enough data for devs.

[Blackbox logs 3 and 4](https://github.com/betaflight/betaflight/files/2334657/20180826_CrashRecoveryBug.zip)
[Video 29s and 1:17s](https://youtu.be/vMxFJv7vrxw?t=28s)
[Diff](https://github.com/si618/quadsquad/blob/master/BetaFlight/Kraken1-BFF4.txt)

Excuse the crap flying, first pack of the day :)

AliB10, 2018-08-30:
Here's another clip of it happening to me.

https://www.youtube.com/watch?v=z--hWbiyOLU&feature=youtu.be

AliB10, 2018-08-30:
I have since turned off crash recovery and no longer get this glitch, so it is related to crash recovery

si618, 2018-08-30:
Yeah for sure, I have turned crash recovery off now as well.  Good thing it's disabled by default!

etracer65, 2018-08-30:
I see an odd behavior in the logs posted above, but not sure yet why it's happening.

Not sure whether it's the problem, but to eliminate one possibility can someone please try:

`set iterm_rotation = OFF`

To see if the behavior changes?  Would be helpful to also capture logs.

etracer65, 2018-08-30:
Never mind on the `term_rotation` test.  I think I may have found the problem related to Iterm Relax.  Unfortunately it can't be fixed by simply turning Iterm Relax off as the code appears to be conflicting with crash recovery.

I have an experimental build that I think should fix the problem.  If you're willing to test, let me know which target name you need and I'll give you a hex file to try.

AliB10, 2018-08-30:
I'll give it a test on OmnibusF4sd

etracer65, 2018-08-30:
@AliB10 Here you go:

[betaflight_3.5.1_OMNIBUSF4SD.hex.zip](https://github.com/betaflight/betaflight/files/2336684/betaflight_3.5.1_OMNIBUSF4SD.hex.zip)

Please try to capture a log with `set debug_mode = CRASH_RECOVERY`.  Also don't forget to re-enable crash recovery.

I test flew this briefly (no crashes) and it seems to be fine.  But since changes were made to the PID controller treat this as highly experimental and follow safety precautions.  Be sure to remove props before connecting a battery and testing arming to ensure nothing unusual happens.  Then for the initial flight be sure to have the quad well away from you arming.  I don't expect any problems but always better to be safe.

AliB10, 2018-08-30:
thanks. Will give it a fly tomorow

si618, 2018-08-31:
Happy to test on BETAFLIGHTF4 target if needed.

etracer65, 2018-08-31:
@si618 Here you go.

[betaflight_3.5.1_BETAFLIGHTF4.hex.zip](https://github.com/betaflight/betaflight/files/2338823/betaflight_3.5.1_BETAFLIGHTF4.hex.zip)

Same thing as earlier - treat as experimental and use caution.  Try to capture a blackbox log using `set debug_mode = CRASH_RECOVERY`.

si618, 2018-09-02:
@etracer65 [Here you go ;-)](https://github.com/betaflight/betaflight/files/2343096/BTFL_BLACKBOX_LOG_SI-FI_20180902.zip)

Wasn't sure how to interpret the debug data, but log 10 seems to have some decent info. I smacked into a tree several times (1:22, 2:14, 2:33) which should've triggered crash recovery, and couldn't reproduce the problem.  

Got some DVR I can upload if you need, just let me know, but looks a lot better, cheers!

etracer65, 2018-09-02:
@si618 A number of good crashes! :)

Looks like this fixes the problem.  Crash Recovery was kicking in and leveling the quad and then shutting back off to return control as it should.  For example, the crash at 1:22:

[image]

The bottom debug graph shows a flag that Crash Recovery is active (debug 3) and debug 0 and debug 1 show the roll and pitch angles respectively.  You can see crash recovery get triggered and the PID controller working to level the quad as the roll/pitch angles move towards 0.  Once they get close then crash recovery shuts off.  The entire process takes about 0.6 seconds.

The problem before was that Iterm Relax got added sort of "in the middle" and was overriding the crash recovery response which was preventing the leveling (or at least slowing it way down).  This then prevented crash recovery from disengaging which explains the effect you saw.  You had control of throttle and yaw but crash recovery was (ineffectively) controlling roll/pitch.  Also because of the way the logic was inserted simply turning off Iterm Relax wouldn't fix the problem either - things needed to be rearranged for a full fix.

Thanks for testing.  I'll get a pull request ready to get the fix merged.

si618, 2018-09-02:
Happy to help. If it's one thing I'm good at, it's crashing :)

etracer65, 2018-09-02:
Created https://github.com/betaflight/betaflight/pull/6682

AliB10, 2018-09-02:
thanks very much. working for me, but didnt have an SD card in so no log but have one in now to check :)

PR #6682 (2018-09-03): Fix iterm relax interfering with crash recovery
Fixes #6477 

Iterm Relax was inserted before crash recovery and this was causing the recovery to be ineffective or fail completely. The problem was that the subsequent iterm calculation was not using the calculated recovery error and instead was using the setpoint error. As such I ended up accumulating and opposing P based on recovery error. This prevented the efforts of crash recovery to level the quad and since it couldn't reach the level state crash recovery wouldn't shut off. This resulted in the pilot only having control over yaw and thrott [... cut]
closed 2018-09-03 by mikeller