betaflight/betaflight #6482: BlackBox Doesn't work on OMNINXT F7

CcLaFleche, 2018-07-29:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

closed 2018-07-29 by jflyper

CcLaFleche, 2018-07-29:
Thanks for the solution !

jflyper, 2018-07-30:
That was the best I could do.

mikeller, 2018-07-30:
@CcLaFleche: If you choose to not supply any of the information that you are asked to provide when opening an issue, please do not be surprised if you don't get any help.