betaflight/betaflight #6488: DalRC F405 Softserial doesn't work

SlipstreamFPV, 2018-07-30:
1. A number of users of this FC can't make softserial work even with a basic configuration. As well as not working the looptime jumps to over 1000 regularly despite 7-10% CPU usage when softserial is enabled. Even with ACC and other features turned off to reduce load.

2. Flash betaflight with complete erase. Turn on softserial, free the uart 2 Tx (A02) pin, assign the A02 pin to softserial. Set soft serial port to SmartPort. Used default and then also changed tlm options to try different settings.

3. DalRC F405 FC connected to Frsky R9 Mini or X4R. SBUS from Rx connected to SBUS pin on FC. S-Port from Rx connected to UART2 Tx pin.

4. Configurator 10.3.1, Betaflight 3.4

5.[diff.txt](https://github.com/betaflight/betaflight/files/2239414/diff.txt)

jflyper, 2018-07-30:
@SlipstreamFPV 

Software serial over PA2 does not work on DALRCF4, as the pin does not have valid timer associated.

Board maintainer should add an appropriate `timerHardware` entry to make this happen. @nyway ?

closed 2018-07-30 by jflyper

SlipstreamFPV, 2018-07-30:
Thanks for the information @jflyper 

Are there any other pins it will work on at the moment?

jflyper, 2018-07-30:
Sec ...

nyway, 2018-07-30:
@SlipstreamFPV 
Hey
DALRCF405 have default pin pad out for vtx_tramp,not necessary use softserial.
If want to use softserial,PB4 maybe good.

jflyper, 2018-07-30:
@SlipstreamFPV 
Only pins/pads eligible for software serial are LED strip (TIM4) and FCCAM (TIM2).
If you are not using them and these are directly exposed, you are lucky.

SlipstreamFPV, 2018-07-30:
Thanks @jflyper 

What a pain this frsky inverted stuff has become. R9 Mini had no uninverted options and most F4 boards... well you know the story /rant

nyway, 2018-07-30:
@jflyper 
Yes,PB4 not wired.But PB4 have TM3_CH1 associated.Or you mean timer should have DMA support?

nyway, 2018-07-30:
@jflyper 
 _If you are a board maintainer, I would suggest that you add backdoor timer channels options to pins assigned for UARTs._
Next hardware version will have this compatibility improvements,thank you.

jflyper, 2018-07-30:
@nyway Refer to OMNINXT dets for backdoors.

jflyper, 2018-07-30:
@nyway No, backdoor timers don't need any hardware mods. Just additional entries in `timerHardware` array.

nyway, 2018-07-30:
@jflyper 
Gotcha.

SlipstreamFPV, 2018-07-30:
Softserial on the LED pin (C14) doesnt work either just as a headsup. Not sure why.

jflyper, 2018-07-30:
LED strip is on PB6.

nyway, 2018-07-30:
@SlipstreamFPV 
It is LED_STRIP PIN,not LED

SlipstreamFPV, 2018-07-30:
@nyway Ah.. skimmed over that too fast. PB6 is not wired out either is it? This FC is not much fun to setup in its current form.

nyway, 2018-07-30:
@SlipstreamFPV 
PB6 wired out.Is LED_STRIP PIN,like pic below:
[image]

SlipstreamFPV, 2018-07-30:
Ah. I messed up. My mistake. Thanks!

jflyper, 2018-07-30:
Looptime jumping over 1000 may be related, but I suspect it's a separate problem.

Can you open a new issue ticket with full `diff`?