betaflight/betaflight #6490: Blackbox log header not complete

McGiverGim, 2018-07-30:
Seeing a Blackbox file, it seems that not all the Header values have been wrote when the rest of the data starts.
In the firmware, these are the latest header elements:
https://github.com/betaflight/betaflight/blob/9e041960479a3b1da991f085236ea1349b786270/src/main/blackbox/blackbox.c#L1327-L1337

But in my Blackbox file this is the end of the Header of a typical log:
[image]

As you can see, the Headers are not complete and the I and P frames starts.

I don't think it matters, but here is the diff of my quad:
```
# diff all defaults

# version
# Betaflight / MATEKF405 (MKF4) 3.4.0 Jul  2 2018 / 09:40:02 (82d761ea3) MSP API: 1.39

board_name MATEKF405
manufacturer_id 
mcu_id 002e00293536471732313130
signature 

# reset configuration to default settings
defaults nosave

# name
name MII-MIGMUL

# resources

# mixer

# servo

# servo mix

# feature
#feature -GPS
feature GPS

# beeper

# beacon

# map

# serial
#serial 0 0 115200 57600 0 115200
serial 0 2 115200 57600 0 115200
#serial 1 64 115200 57600 0 115200
serial 1 4160 115200 57600 0 115200
#serial 3 0 115200 57600 0 115200
serial 3 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
#aux 0 0 0 900 900 0
aux 0 0 0 1950 2050 0
#aux 1 0 0 900 900 0
aux 1 1 1 1250 1350 0
#aux 2 0 0 900 900 0
aux 2 1 1 1450 1550 0
#aux 3 0 0 900 900 0
aux 3 1 1 1650 1850 0
#aux 4 0 0 900 900 0
aux 4 2 1 1350 1450 0
#aux 5 0 0 900 900 0
aux 5 3 1 1250 1350 0
#aux 6 0 0 900 900 0
aux 6 3 1 1650 1850 0
#aux 7 0 0 900 900 0
aux 7 6 1 1750 1850 0
#aux 8 0 0 900 900 0
aux 8 10 1 1250 1350 0
#aux 9 0 0 900 900 0
aux 9 11 1 1250 1350 0
#aux 10 0 0 900 900 0
aux 10 11 1 1650 1850 0
#aux 11 0 0 900 900 0
aux 11 13 1 1250 1350 0
#aux 12 0 0 900 900 0
aux 12 26 0 1950 2050 0
#aux 13 0 0 900 900 0
aux 13 28 1 1050 1150 0
#aux 14 0 0 900 900 0
aux 14 32 5 1950 2050 0
#aux 15 0 0 900 900 0
aux 15 33 4 950 1200 0
#aux 16 0 0 900 900 0
aux 16 34 4 1800 2050 0

# adjrange
#adjrange 0 0 0 900 900 0 0 0 0
adjrange 0 0 2 1800 2050 17 3 0 0
#adjrange 1 0 0 900 900 0 0 0 0
adjrange 1 0 2 950 1200 15 3 0 0

# rxrange

# vtx

# rxfail
#rxfail 5 h 1500
rxfail 5 s 1500

# master
#set gyro_lowpass_hz = 100
set gyro_lowpass_hz = 90
#set gyro_lowpass2_hz = 300
set gyro_lowpass2_hz = 0
#set acc_calibration = 0,0,0
set acc_calibration = 19,62,-92
#set align_mag = DEFAULT
set align_mag = CW270FLIP
#set mag_declination = 0
set mag_declination = -110
#set mag_calibration = 0,0,0
set mag_calibration = -338,-516,-628
#set max_check = 1900
set max_check = 1950
#set rc_smoothing_type = INTERPOLATION
set rc_smoothing_type = FILTER
#set serialrx_provider = SBUS
set serialrx_provider = IBUS
#set blackbox_p_ratio = 32
set blackbox_p_ratio = 64
#set dshot_idle_value = 450
set dshot_idle_value = 550
#set motor_pwm_protocol = ONESHOT125
set motor_pwm_protocol = DSHOT600
#set failsafe_delay = 4
set failsafe_delay = 3
#set failsafe_off_delay = 10
set failsafe_off_delay = 20
#set failsafe_throttle = 1000
set failsafe_throttle = 1300
#set failsafe_procedure = DROP
set failsafe_procedure = AUTO-LAND
#set bat_capacity = 0
set bat_capacity = 1800
#set vbat_detect_cell_voltage = 30
set vbat_detect_cell_voltage = 55
#set beeper_dshot_beacon_tone = 1
set beeper_dshot_beacon_tone = 2
#set gps_provider = NMEA
set gps_provider = UBLOX
#set gps_sbas_mode = AUTO
set gps_sbas_mode = EGNOS
#set gps_ublox_use_galileo = OFF
set gps_ublox_use_galileo = ON
#set deadband = 0
set deadband = 3
#set yaw_deadband = 0
set yaw_deadband = 5
#set pid_process_denom = 4
set pid_process_denom = 1
#set osd_rssi_alarm = 20
set osd_rssi_alarm = 0
#set osd_cap_alarm = 2200
set osd_cap_alarm = 1600
#set osd_tim1 = 2560
set osd_tim1 = 0
#set osd_tim2 = 2561
set osd_tim2 = 1793
#set osd_vbat_pos = 234
set osd_vbat_pos = 2433
#set osd_rssi_pos = 234
set osd_rssi_pos = 46
#set osd_tim_1_pos = 234
set osd_tim_1_pos = 54
#set osd_tim_2_pos = 234
set osd_tim_2_pos = 2455
#set osd_flymode_pos = 234
set osd_flymode_pos = 2145
#set osd_throttle_pos = 234
set osd_throttle_pos = 2424
#set osd_vtx_channel_pos = 234
set osd_vtx_channel_pos = 376
#set osd_crosshairs_pos = 205
set osd_crosshairs_pos = 2253
#set osd_ah_sbar_pos = 206
set osd_ah_sbar_pos = 2254
#set osd_ah_pos = 78
set osd_ah_pos = 2126
#set osd_current_pos = 234
set osd_current_pos = 2400
#set osd_mah_drawn_pos = 234
set osd_mah_drawn_pos = 2370
#set osd_craft_name_pos = 234
set osd_craft_name_pos = 2442
#set osd_gps_speed_pos = 234
set osd_gps_speed_pos = 2262
#set osd_gps_lon_pos = 234
set osd_gps_lon_pos = 2129
#set osd_gps_lat_pos = 234
set osd_gps_lat_pos = 2097
#set osd_gps_sats_pos = 234
set osd_gps_sats_pos = 2120
#set osd_home_dir_pos = 234
set osd_home_dir_pos = 2126
#set osd_home_dist_pos = 234
set osd_home_dist_pos = 2090
#set osd_compass_bar_pos = 234
set osd_compass_bar_pos = 2154
#set osd_altitude_pos = 234
set osd_altitude_pos = 2294
#set osd_pid_roll_pos = 234
set osd_pid_roll_pos = 359
#set osd_pid_pitch_pos = 234
set osd_pid_pitch_pos = 359
#set osd_pid_yaw_pos = 234
set osd_pid_yaw_pos = 359
#set osd_debug_pos = 234
set osd_debug_pos = 391
#set osd_power_pos = 234
set osd_power_pos = 321
#set osd_pidrate_profile_pos = 234
set osd_pidrate_profile_pos = 344
#set osd_warnings_pos = 2377
set osd_warnings_pos = 2345
#set osd_avg_cell_voltage_pos = 234
set osd_avg_cell_voltage_pos = 76
#set osd_pit_ang_pos = 234
set osd_pit_ang_pos = 257
#set osd_rol_ang_pos = 234
set osd_rol_ang_pos = 289
#set osd_battery_usage_pos = 234
set osd_battery_usage_pos = 302
#set osd_disarmed_pos = 234
set osd_disarmed_pos = 2379
#set osd_nheading_pos = 234
set osd_nheading_pos = 106
#set osd_nvario_pos = 234
set osd_nvario_pos = 1
#set osd_esc_tmp_pos = 234
set osd_esc_tmp_pos = 82
#set osd_esc_rpm_pos = 234
set osd_esc_rpm_pos = 83
#set osd_stat_rtc_date_time = OFF
set osd_stat_rtc_date_time = ON
#set osd_stat_max_dist = OFF
set osd_stat_max_dist = ON
#set osd_stat_min_batt = ON
set osd_stat_min_batt = OFF
#set osd_stat_endbatt = OFF
set osd_stat_endbatt = ON
#set osd_stat_min_rssi = ON
set osd_stat_min_rssi = OFF
#set osd_stat_max_alt = OFF
set osd_stat_max_alt = ON
#set debug_mode = NONE
set debug_mode = RC_SMOOTHING
#set vcd_video_system = AUTO
set vcd_video_system = NTSC

# profile
profile 0

#set anti_gravity_gain = 5000
set anti_gravity_gain = 6000
#set iterm_rotation = ON
set iterm_rotation = OFF
#set iterm_relax = OFF
set iterm_relax = RP
#set iterm_relax_type = GYRO
set iterm_relax_type = SETPOINT
#set p_pitch = 50
set p_pitch = 45
#set i_pitch = 50
set i_pitch = 60
#set d_pitch = 27
set d_pitch = 47
#set f_pitch = 60
set f_pitch = 70
#set p_roll = 46
set p_roll = 41
#set d_roll = 25
set d_roll = 38
#set f_roll = 65
set f_roll = 75
#set p_yaw = 65
set p_yaw = 80
#set i_yaw = 45
set i_yaw = 80
#set f_yaw = 60
set f_yaw = 70

# profile
profile 1

#set dterm_notch_cutoff = 160
set dterm_notch_cutoff = 0
#set anti_gravity_gain = 5000
set anti_gravity_gain = 6000
#set iterm_rotation = ON
set iterm_rotation = OFF
#set iterm_relax = OFF
set iterm_relax = RP
#set iterm_relax_type = GYRO
set iterm_relax_type = SETPOINT
#set p_pitch = 50
set p_pitch = 52
#set i_pitch = 50
set i_pitch = 70
#set d_pitch = 27
set d_pitch = 40
#set f_pitch = 60
set f_pitch = 75
#set p_roll = 46
set p_roll = 41
#set d_roll = 25
set d_roll = 35
#set f_roll = 65
set f_roll = 80
#set f_yaw = 60
set f_yaw = 80

# profile
profile 2

#set iterm_rotation = ON
set iterm_rotation = OFF

# restore original profile selection
profile 1

# rateprofile
rateprofile 0

#set thr_mid = 50
set thr_mid = 40
#set roll_expo = 0
set roll_expo = 20
#set pitch_expo = 0
set pitch_expo = 20
#set roll_srate = 70
set roll_srate = 75
#set pitch_srate = 70
set pitch_srate = 71
#set yaw_srate = 70
set yaw_srate = 61
#set tpa_rate = 10
set tpa_rate = 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

McGiverGim, 2018-07-31:
@etracer65 I was thinking... 
- Your log is from integrated memory or from sdcard?
- What rate are you logging the sdcard?

Maybe here are the differences.

etracer65, 2018-07-31:
The screenshot above is from on-board dataflash.  But I just tested with SD card and it was fine also.  I was logging at 2KHz.

McGiverGim, 2018-08-06:
@etracer65 I continue with the same bug. Can you attach me one of your Blackbox files as example to see if I can figure the difference? Thanks!

etracer65, 2018-08-06:
@McGiverGim I was able to replicate the problem using your `diff` above (on a OMNIBUSF4SD).  My current theory is that with lots of extra features enabled we might be overflowing a buffer used to output the header.

McGiverGim, 2018-08-06:
That's good! I'm not the only one and you have confirmed the bug. Thanks!

etracer65, 2018-08-06:
Well not sure yet.  Was only able to replicate it twice with a bunch of times where no problem was seen.  So difficult to debug...

McGiverGim, 2018-08-06:
I suppose the header is written at some moment, I've it configured when armed. At the same time, in the pid loop, the blackbox is called to write the data. If the data is written to early, maybe the header has not finished and this problem occurs.

This is only my supposition.

mikeller, 2018-08-07:
@McGiverGim: There is no 'at the same time' in Betaflight (except in the cases of the few ISRs that we have, for example on serial input), since time sharing is done with collaborative multitasking: Every task is allowed to run until it voluntarily returns control, and then the next task is run.

etracer65, 2018-08-07:
I've been testing and (I think) I've nailed it down to some problem with the cached blocks not flushing (fast enough?) for SD card.  What I've found so far:

- Problem only happens with SD card. FlashFS doesn't have a problem. Can't test serial.
- The problem is definitely some kind of overflow or buffer overrun when in the `BLACKBOX_STATE_SEND_SYSINFO` state that writes the informational headers.
- Rearranging the fields (or adding more) to `blackboxWriteSysinfo()` causes the corruption to appear at different endpoints, but always roughly after the same number of fields (bytes).  So it's not some particular field that's the problem, but how much data is written.

I think it's related to the cached blocks not being fully flushed and then the subsequent blackbox logging fields cause an overrun condition.  Related to:
```
        if (blackboxWriteSysinfo()) {
            /*
             * Wait for header buffers to drain completely before data logging begins to ensure reliable header delivery
             * (overflowing circular buffers causes all data to be discarded, so the first few logged iterations
             * could wipe out the end of the header if we weren't careful)
             */
            if (blackboxDeviceFlushForce()) {
                blackboxSetState(BLACKBOX_STATE_RUNNING);
            }
        }
```
So after the last header field is written `blackboxWriteSysinfo()` returns true which forces the flush and advances the blockbox logging state.

In testing I found that increasing the block cache size `AFATFS_NUM_CACHE_SECTORS` from 8 to 16 prevents the problem.  This is not the solution I think but just helps confirm the flush/overrun problem.

The function `blackboxDeviceFlushForce()` just ends up calling `afatfs_flush()` which seems to write the dirty blocks but has a deferred callback type of operation that marks the block as flushed (from an upstream perspective) however they may not actually be written yet.  So I'm wondering if the actual block writes in the SD card are just too slow (sometimes) and the block cache is filling up with the in-process writes and new data from the actual logging.

I really don't have any experience with the SD card logic or the fatfs emulation so I'm not sure I can go much further.

I don't think this is related to any problems in 3.5 and is a problem that's likely been around for a while.  It might also explain why occasionally I've seen intermittent cases of empty blackbox logs on SD cards.  I've had cases where I have a bench test setup where I might capture a bunch of logs back to back with different settings during testing.  Inexplicably sometimes one right in the middle will be created with 0 bytes even though I know a log should have been generated.  Might be another instance of the block cache being overrun preventing the file from being created properly - just a guess.

DieHertz, 2018-08-07:
There's also no "at the same time" as there's only one physical thread of execution.

ombracol, 2018-08-08:
My sd wont log at all..I get log number in osd but no logs on sd

etracer65, 2018-08-08:
Different issue. Try reformatting your SD card.

flint723, 2018-08-08:
I've had lots of problems with sd card logging on the SPRacing F7 Dual as well.  Times where header is incomplete (lots of red boxes), or the log is somehow corrupted and causes BBE to freeze after trying to scroll around.  Seems a bit better lately (maybe because I am not pushing the CPU as much).  Logging at 2khz also seems a bit worse than 1.5khz.

ombracol, 2018-08-08:
I will look for any open issues or open one  ..but its happening on my other board as well  that uses on board  flash.

McGiverGim, 2018-08-08:
It seems that my problem yesterday with the two files with zero bytes was something wrong with the SDCard. It get corrupted. I needed to format it to make it work again. I don't know if the corruption was caused by some problem or it simply was the typical SDCard reliability.

0crap, 2018-08-08:
It seems I run into the same issue.

[image]

Target is Matek F405-STD. Using the on board SD card slot for logging.
BF 3.5.0-RC1 and 3.5.0-RC2 have this issue.
At 4KHz log rate almost all my log files are useless, sometimes a good one.
At 2KHz log rate it works better, but sometimes a useless one.

ralenekov, 2018-08-08:
https://github.com/betaflight/betaflight/issues/5946#issuecomment-392506954 I can reproduce this easily as well with early jenkins builds of 3.4 till 3.5-rc2. Using Matek F405-STD as well. 8k/8k, 32k/16k, DSHOT1200/DSHOT600, it doesn't matter (almost default settings). Tried another sdcard, it doesn't matter as well.

Workaround 1:
If you copy the headers from a working log to a corrupted log, it works as expected in the log viewer.

Workaround 2:
When in flight, start and log several logs, which increases the chance of having a working one, without the need to fix it in a text editor.

mikeller, 2018-08-08:
There is one change that we did pre-3.4, and that was moving the periodical blackbox-flush calls from being tied to the PID loop to being called periodically: #6126 
I think this could have an influence that would ultimately cause this, or exacerbate it.

@ombracol: Try formatting your card with this tool: https://www.sdcard.org/downloads/formatter_4/index.html

etracer65, 2018-08-08:
@mikeller #6126 might be partially related to other BB log problems (dropped frames or maybe empty files), but I'm not sure it's directly related to this problem.  The reason being is that after the headers a written the code tries to ensure a full flush of all dirty cache blocks before starting the actual logging.  The only other place it does this is on shutdown.  The way the logic works is that the dirty blocks are likely getting a deferred writing state while waiting for the actual block writes to occur and the callback that flags them as in-sync.  This deferred writing state looks like there are no longer any dirty blocks and the logging then proceeds but the cache entries are still used - probably causing the cache to overflow.  Increasing the cache size made things work so this indirectly helps to confirm.

So I'm going to try and change the logic some to ensure the cache is truly written (deferred writes complete) before allowing the logging to start writing "I" and "P" frames to see if that helps.

0crap, 2018-08-16:
Would be nice if this one, when solved, can go into version 3.5.1

mikeller, 2018-08-16:
@0crap: Bugfixes normally go into patch releases (like 3.5.1) - that's what they are there for.

McGiverGim, 2018-10-25:
Only to confirm that tested the latest master and the bug still remains.

McGiverGim, 2019-05-06:
The bug still remains. In the Slack chat it seems other people is suffering it. Maybe the header is growing so this bug will be worse in time.

etracer65, 2019-05-06:
I've looked at this a number of times and haven't been able to find a cause. It also is difficult to replicate (at least for me). I suspected that it's some kind of buffering/flush bug but that's all I have so far.

etracer65, 2019-05-16:
@McGiverGim I think finally found and fixed the problem. Can you try #8284 ?

0crap, 2019-05-16:
That would be totally awesome!
Such a nasty bug this one.
3 out of 10 logs I can not use.

[image]

McGiverGim, 2019-05-16:
Thanks @etracer65 for your work at fixing it!!! I will test it, but the problem is random, so I can't be sure that it is fixed or not. I don't know why but lately the bug only appears from time to time.

etracer65, 2019-05-16:
Note that this only addresses SD card logging - which is the only way I could replicate. When testing with on-board flash (which uses a different filesystem) was not able to replicate the problem.

0crap, 2019-05-16:
Yep, I'm logging on SD card only.

McGiverGim, 2019-05-16:
Me too. Only SDCard.

etracer65, 2019-05-16:
@McGiverGim It has to do with how much data is written to the header. And this varies depending on settings. As soon as it tries to write 3073 bytes (3K + 1) we have the problem. If you want to force the problem then just modify `blackbox.c` and replicate some of the header write lines to add more to the header. Like replicate the entire `USE_RC_SMOOTHING_FILTER` section a couple of extra times.

McGiverGim, 2019-05-17:
I have logs with a header greater than 3073 that worked before the patch. Maybe I'm not sizing it right. 

If it matters I have generated a lot blackbox files with the patch and all of them are correct.

PR #8284 (2019-05-18): Increase async FAT filesystem cache sectors to prevent blackbox header corruption
Fixes #6490 

In testing I found that once the blackbox header exceeded 3072 bytes (3K) it would be truncated and overwritten with logging data. 3K would be 6 sectors and it looks like when it tried to flow over into 7 sectors the previous cache of 8 sectors was insufficient when considering the other FAT filesystem overhead. Increased the cache to 10 sectors (1K extra memory usage) and the problem is resolved. If the blackbox header continues to expand in size this will have to be revisited.
closed 2019-05-18 by mikeller