betaflight/betaflight #6492: Cant arm in Betaflight.

Gissa2205, 2018-07-30:
Can't arm Quad. In BF unless having the usb hooked to FC. and arming motors through BF motors app. than just disconnect USB while motor tab still turnd on than can arm using Not being able arm quad ib BF unless Trick BF or FC. 
Steps to reproduce the behavior:To recreate Have Quad all ready too set up in BF using BF10.3.1 Connect USB set up everything but can't arm Quad without turning on motors in BF app. than Without exiting BF just disconnect USB and can arm Quad normally .

**Expected behavior**I expect it to arm like in many other Quads I have built.
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

varkokonyi, 2018-07-30:
17 is just the MSP connection being active, completely nominal. For the other flags, check out https://github.com/betaflight/betaflight/wiki/Arming-Sequence-&-Safety
I'm also suspecting a hardware failure, can you check a different version?

etracer65, 2018-07-30:
Not exactly sure what you're trying to say.  Are you saying you can't arm while connected to the configurator?  If so that's normal and is a safety feature.  If you remove the props and want to test in the configurator then you need to enable the safety switch on the motors tab - then you can arm.

This is a restriction only when connected to the configurator.  Try disconnecting the USB cable with arming with a battery powering your quad.

Arming disable reason 3 is `RX LOSS` so make sure you have your TX/RX bound and your transmitter powered on.  Test that the channels move in the Receiver tab in the configurator.

mikeller, 2018-07-31:
@Gissa2205: Having the output of a `diff` like you were asked to provide when opening this issue would really help to enable others to be able to diagnose your problem...

JustinBluntFPV, 2018-08-10:
i have the same problem. it wont arm, except after i plug it into betaflight and then exit and dont power cycle. ive used the latest version of 3.5 and now 3.4.1 and i have a matek f722 FC. i have my DIFF  do i just paste it all?

# diff

# version
# Betaflight / MATEKF722 (MKF7) 3.4.1 Aug  2 2018 / 08:11:51 (1d99444d9) MSP API: 1.39

board_name MATEKF722
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 13 3 1700 2100 0
aux 4 28 2 1300 1700 0
aux 5 30 2 1300 1700 0
aux 6 47 2 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_use_32khz = ON
set mag_hardware = NONE
set min_check = 1100
set rc_interp_ch = RPYT
set min_throttle = 1070
set dshot_idle_value = 400
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set current_meter = NONE
set battery_meter = NONE
set small_angle = 180
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2081
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_altitude_pos = 2113
set osd_disarmed_pos = 2282
set osd_core_temp_pos = 2136
set cpu_overclock = 240MHZ
set vcd_video_system = NTSC
set max7456_clock = FULL

# profile
profile 0

set anti_gravity_threshold = 250

# rateprofile
rateprofile 0

closed 2018-09-16 by stale[bot]