betaflight/betaflight #6495: Unstable Looptime with softserial enabled (DALRC F405)

SlipstreamFPV, 2018-07-30:
Once soft serial is enabled on the DALRC F405 the looptime becomes ocassionally unstable even though the CPU usage is in the single digit percentages. The looptime will jump over 1000 sometimes.

Reproduce: Flash betaflight with defaults, enable some features including softserial.

Expected: Stable looptime 

FC Config:
```

# diff

# version
# Betaflight / DALRCF405 (DLF4) 3.4.0 Jul  8 2018 / 04:07:08 (cce8a59be) MSP API: 1.39

board_name DALRCF405
manufacturer_id 

# name

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B06

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature OSD
feature AIRMODE

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_hardware = NONE
set mag_hardware = NONE
set motor_pwm_protocol = DSHOT1200

# profile
profile 0

# rateprofile
rateprofile 0

```
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Setup/Versions:
 - DALRC F405
 - Fault reproducable without any additional hardware (FC plugged into USB only)

Additional:
Happens on both DALRC F405 and DALRC F405 AIO.

jflyper, 2018-07-31:
@SlipstreamFPV Thanks for opening a new issue ticket.

Which acc/gyro chip do you have (MPU or ICM)?

etracer65, 2018-07-31:
It's not completely clear, but are you trying to use softserial for serial RX?  If so that's definitely not recommended.

SlipstreamFPV, 2018-07-31:
No. I’ve only used it for smartport telemetry.

jflyper, 2018-07-31:
@SlipstreamFPV 
First, a couple of questions about your configuration.
- Your `diff` looks strange, as it does not configure a serial receiver (`-RX_SERIAL`).
- It also does not show the software serial configured for S.Port.

Then another couple about operational status.
- Is the S.Port working?
- Does the symptom develops without S.Port signal (not connected or receiver not powered)?

SlipstreamFPV, 2018-07-31:
Ah, the diff is from a spare board I had. I did just enough setup changes to make the board have the looptime glitch. It glitches without any other hardware actually attached. 

One my other quad with the same board it’s running sbus (via normal serial port) and sport (via soft serial) and has the same problem with hardware attached. Rx is r9 mini.

jflyper, 2018-07-31:
@SlipstreamFPV 
Can we see your `diff` output, please?

And you are saying the symptom develops as soon as you apply this diff to a bare board, without anything connected?

SlipstreamFPV, 2018-07-31:
Yes, I’m saying with no other hardware connected and the diff above it happens (which is why I posted that diff. I wouldn’t have thought anything on top of that would be worthwhile).

Here is my built quad, same problem:

```
# diff

# version
# Betaflight / DALRCF405 (DLF4) 3.4.0 Jul  8 2018 / 04:07:08 (cce8a59be) MSP API: 1.39

board_name DALRCF405
manufacturer_id 

# resources
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature OSD
feature AIRMODE

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2100 0
aux 1 13 2 1925 2100 0
aux 2 35 1 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set mag_hardware = NONE
set rssi_channel = 8
set rc_smoothing_type = FILTER
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT1200
set ibata_scale = 210
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5645
set camera_control_ref_voltage = 320
set camera_control_internal_resistance = 460

# profile
profile 0

set dterm_notch_cutoff = 0
set dterm_setpoint_weight = 85
set iterm_relax = RP
set i_pitch = 66
set i_roll = 55

# rateprofile
rateprofile 0

set roll_srate = 78
set pitch_srate = 78
```

jflyper, 2018-08-07:
Can't reproduce with generic board. Someone with the board should test this diff output for reproduction. @nyway ?

nyway, 2018-08-07:
@jflyper 
I'll test it,wait a minute.

nyway, 2018-08-07:
@jflyper   @SlipstreamFPV  

This is my diff.
About the looptime,did you mean that was**Cycle Time**?
[image]

```
diff

# version
# Betaflight / DALRCF405 (DLF4) 3.4.1 Aug  2 2018 / 07:52:31 (1d99444d9) MSP API: 1.39

board_name DALRCF405
manufacturer_id 

# name

# resources
resource SERIAL_TX 11 B04

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD
feature AIRMODE

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set motor_pwm_protocol = DSHOT1200

# profile
profile 0

# rateprofile
rateprofile 0

# 
```

jflyper, 2018-08-07:
@nyway 
Yeah, it is claimed that the cycle time fluctuates.
Can you `defaults novae` then paste `diff` output from this comment and test?
https://github.com/betaflight/betaflight/issues/6495#issuecomment-409170787

nyway, 2018-08-07:
@jflyper  @SlipstreamFPV 
Now same as SlipstreamFPV's diff.
Cycle time about 251,sometimg quickly turn 1000 then back to 251.
When i set Gyro update frequency 8K,PID loop frequency 2K,Cycle time turn back about 127.
If dual 2K/2K,Cycle time 500,dual 1K/1K,Cycle time about 1000.
I full erase and reflash it,and set nothing except Gyro update frequency and PID loop frequency 1K/1K, Cycle time about 1000 too.
Have nothing to do with softserial,maybe just PID utilization,just my guess.
And not just DALRCF405 target,maybe others too i think.
[image]

```
# diff

# version
# Betaflight / DALRCF405 (DLF4) 3.4.1 Aug  2 2018 / 07:52:31 (1d99444d9) MSP API: 1.39

board_name DALRCF405
manufacturer_id 

# name

# resources
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature OSD
feature AIRMODE

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2100 0
aux 1 13 2 1925 2100 0
aux 2 35 1 1300 1700 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set mag_hardware = NONE
set rssi_channel = 8
set rc_smoothing_type = FILTER
set blackbox_p_ratio = 128
set motor_pwm_protocol = DSHOT1200
set ibata_scale = 210
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5645
set camera_control_ref_voltage = 320
set camera_control_internal_resistance = 460

# profile
profile 0

set dterm_notch_cutoff = 0
set dterm_setpoint_weight = 85
set iterm_relax = RP
set i_pitch = 66
set i_roll = 55

# rateprofile
rateprofile 0

set roll_srate = 78
set pitch_srate = 78
```

closed 2018-09-13 by stale[bot]