betaflight/betaflight #6499: Throttle boost value can't be changed via osd

Hydroculture, 2018-07-31:
Throttle boost is shown on osd, but value can't be changed.
FC is HGLRC F440 (omnibusf4)

Thank you

best Holger

etracer65, 2018-07-31:
Well since you provided almost none of the information requested it's difficult to answer your question.  At least provide details of what version of Betaflight you are using.

It looks like the item you're describing was just added to the code last night so unless you are running a current developer build I wouldn't expect it to work.

See: https://github.com/betaflight/betaflight/pull/6496

Hydroculture, 2018-08-01:
Hello etracer56!
Thank you for your replay.  Sorry for beiing so unprecise.
Yes I installed http://andwho.sytes.net:8080/job/BorisB_BetaFlight/2431/changes, which adds osd support for throttleboost for the fiirst time.
I can enter OSD and change values. Throttle boost is also shown in OSD,  but value cant be changed via OSD.

Thamk you so much  for your incredible work.

best
Holger

m013, 2018-08-01:
I see that as well.
I set trottle boost to 25 via cli, but in osd it is shown something as 547 and could not be chaged

Hakrc f4

McGiverGim, 2018-08-01:
Maybe I added something wrong. I will check it.

McGiverGim, 2018-08-01:
Problem found, I will push the PR in 5 minutes, testing it now...

Hydroculture, 2018-08-01:
Thank you very, very much!

best
Holger

closed 2018-08-01 by Hydroculture