betaflight/betaflight #6500: sp f7 unable to use uart 4 and 5 at the same time

Gunaveh, 2018-07-31:
**Describe the bug**
FC hangs when uart 4 and 5 are enabled with smart audio and runcam device respectively.

**To Reproduce**
Enabling TBS smart audio on uart 4 and runcam device on uart 5 creates a fault where the configurator allows a connection, but no parameters can be changed. The splash screen on the osd is also stuck on.

Uninstalling the vtx board allows me to connect and disable uart 5. CLI is unresponsive with uart 5 enabled.

**Flight controller configuration**
```# diff

# version
# Betaflight / SPRACINGF7DUAL (SP7D) 3.4.0 Jul  8 2018 / 04:34:20 (cce8a59be) MSP API: 1.39

board_name SPRACINGF7DUAL
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -TELEMETRY
feature -RSSI_ADC
feature -LED_STRIP
feature -TRANSPONDER

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 4 0 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set mag_hardware = NONE
set baro_hardware = NONE
set vtx_channel = 2
set vtx_freq = 5760

# profile
profile 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
 - Sp racing f7 dual with vtx board
 - xm+, tbs unify pro v3
 - xm+ and unify direct soldered to vtx board

seven2099, 2018-08-08:
Hello,

I have the exact same issue occurring.
Using Betaflight 3.5 and eventually after resetting, the flight controller hardresets and my settings are erased.

Pretty massive bug.

DieHertz, 2018-08-08:
Will look into it this weekend

closed 2018-09-30 by stale[bot]