betaflight/betaflight #6502: AKK VTX (SmartAudio) extremely weak signal on boot

darmach, 2018-08-01:
VTX despite showing E:3:4 (max power) on OSD seem to send an extremely weak signal (range about 50 cm). When I switch VTX power in OSD (SmartAudio - AKK) to 500 the signal is fixed, I can then switch back to 800 without the issues.

I noticed there are AKK Patch branches of 3.3 and 3.4 firmware, but I was unable my issue in their description.
Did anyone else encountered something like this?

Config diff:
```
# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.3.2 May  4 2018 / 09:52:25 (64a292f37) MSP API: 1.38

# name

# resources

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE
feature ANTI_GRAVITY
feature DYNAMIC_FILTER

# beeper
beeper -ON_USB

# map

# serial
serial 2 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led
led 0 6,15::CW:5
led 1 9,15::CW:5

# color

# mode_color

# aux
aux 0 0 2 1925 2100 0
aux 1 1 0 900 1300 0
aux 2 2 0 1300 1700 0
aux 3 3 1 1300 2100 0
aux 4 27 3 1700 2100 0
aux 5 13 3 1300 2100 0
aux 6 19 5 1750 2100 0
aux 7 26 2 1900 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_notch1_hz = 0
set gyro_notch2_hz = 0
set acc_trim_pitch = 2
set acc_trim_roll = 10
set min_check = 1200
set max_check = 1800
set serialrx_provider = IBUS
set rx_min_usec = 950
set rx_max_usec = 2000
set blackbox_device = SPIFLASH
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_off_delay = 30
set failsafe_throttle = 1050
set align_board_yaw = 180
set bat_capacity = 1300
set vbat_scale = 111
set ibata_scale = 480
set small_angle = 30
set pid_process_denom = 1
set osd_warnings = 63
set osd_cap_alarm = 1100
set osd_vbat_pos = 2497
set osd_tim_1_pos = 2518
set osd_tim_2_pos = 2486
set osd_flymode_pos = 2081
set osd_throttle_pos = 2465
set osd_vtx_channel_pos = 2508
set osd_current_pos = 2272
set osd_mah_drawn_pos = 2241
set osd_altitude_pos = 2294
set osd_power_pos = 2305
set osd_warnings_pos = 2409
set osd_disarmed_pos = 2154
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set vtx_band = 3
set vtx_channel = 3
set vtx_power = 4
set vtx_freq = 5665
set vcd_video_system = 1

# profile
profile 2

# rateprofile
rateprofile 0

set roll_rc_rate = 70
set pitch_rc_rate = 70
set yaw_rc_rate = 70

```

McGiverGim, 2018-08-01:
Maybe is in pit mode? https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio

darmach, 2018-08-01:
> I can't say, but you can verify it in the OSD, I think that the VTX menu
> indicates the mode.

Yes, thanks for the advice - I'll verify it when I get home .) Hopefully
that's just that

darmach, 2018-08-02:
@McGiverGim So lame... You were right! I had VTX configured for Race mode, with PIR.

closed 2018-08-02 by darmach