betaflight/betaflight #6504: Date and time from Tarnis RTC not showing on OSD or POST

FasaArg, 2018-08-01:
**Describe the bug**
I´m not getting date and time from my taranis RTC on the OSD, the the live feed do not show it and post flight statics show NO RTC.
Right now i´m running LUA scripts (1.0.2) and is all working except for the PIT mode that I can not turn off. (#6312) but i can change band, channels, pid, and any other value.
In the same way, i have all the telemetry working with alerts and announces and all work ok.
Just updated my Taranis from 2.2.0 to 2.2.2 and both problems persist.
I don´t know if the status command should show the date and time in the CLI when working, but nothing appears in there.

**To Reproduce**
With both options configured in the OSD and none of them shows up.

**Expected behavior**
Date and time shown as configured.

**Flight controller configuration**
```# diff

# version
# Betaflight / KAKUTEF4 (KTV1) 3.4.0 Jul  8 2018 / 04:15:41 (cce8a59be) MSP API: 1.39

board_name KAKUTEF4
manufacturer_id 

# name
name Fasa

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A01

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature AIRMODE

# beeper
beeper -ON_USB

# beacon

# map

# serial
serial 5 16384 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1950 2050 0
aux 1 1 1 950 1050 0
aux 2 2 1 1450 1550 0
aux 3 13 2 1950 2050 0
aux 4 33 3 1950 2050 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 5 s 1000
rxfail 6 s 2000
rxfail 7 s 1000

# master
set gyro_lowpass_hz = 120
set acc_calibration = 60,36,35
set min_check = 1005
set max_check = 1995
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set rx_min_usec = 950
set rx_max_usec = 2050
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 975
set vbat_min_cell_voltage = 35
set vbat_warning_cell_voltage = 36
set current_meter = ADC
set ibata_scale = 316
set ibata_offset = -3
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_tim2 = 1025
set osd_vbat_pos = 361
set osd_rssi_pos = 352
set osd_tim_1_pos = 321
set osd_tim_2_pos = 2433
set osd_flymode_pos = 3
set osd_throttle_pos = 376
set osd_vtx_channel_pos = 364
set osd_crosshairs_pos = 2253
set osd_ah_pos = 200
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2448
set osd_craft_name_pos = 23
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 42
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 290
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2441
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 359
set osd_disarmed_pos = 2155
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_rtc_date_time_pos = 6
set osd_stat_rtc_date_time = ON
set osd_stat_max_spd = OFF
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 3
set vtx_freq = 5695
set vcd_video_system = NTSC
set vcd_v_offset = -5
set displayport_max7456_col_adjust = -2

# profile
profile 2

set dterm_lowpass2_hz = 250
set dterm_notch_cutoff = 0
set setpoint_relax_ratio = 50
set dterm_setpoint_weight = 80
set iterm_relax = RP
set iterm_relax_cutoff = 10
set p_pitch = 58
set i_pitch = 65
set d_pitch = 35
set p_roll = 50
set i_roll = 60
set d_roll = 30
set i_yaw = 60
set d_yaw = 20

# rateprofile
rateprofile 0

set roll_rc_rate = 122
set pitch_rc_rate = 122
set yaw_rc_rate = 120
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set tpa_rate = 15
set tpa_breakpoint = 1050
```

**Setup / Versions**
 - Flight controller: 
Kakute F4 AIO running BF 3.4.0 (Final)

 - Other components:
RX: FrSky R-XSR fw:171009 (SMARTPORT+SBUS)
VTX: TBS Unify PRO V2 (using SmartAudio via SOFTSERIAL) LED strip disabled as shown in the DIFF
Camera: Runcam Split 2 fw:2.0.0 (control via UART6)
Transmitter: FrSky Taranis X9D+ SE with OpenTX 2.2.2 (same problem on 2.2.0)

**Additional context**
NA

FasaArg, 2018-08-03:
Same behavior on 3.4.1

seaweeduk, 2018-08-06:
RTC is working for me in my R-XSR setups (also 171009 firmware), but one thing I have noticed is that it only works if the radio is powered on after the quad. If I change batteries in the quad and don't power cycle my Taranis it doesn't gets the date. 

Were you switching the radio on after the quad in your tests?

mikeller, 2018-08-07:
There is a change coming in OpenTx 2.2.2, allowing the RTC to be set in milliseconds, instead of YMDhms, once this is out I will look into changing this over, and debugging why the RTC transfer does not work reliably.

seaweeduk, 2018-08-07:
Awesome, many thanks mike looking forward to it!

seaweeduk, 2018-08-08:
@FasaArg my Taranis is still on OpenTx 2.2.1, you might want to test downgrading to this version in case it helps with your problem

mikeller, 2018-09-04:
@seaweeduk: Tested latest unreleased lua scripts with latest unreleased firmware, and this is working now.

FasaArg, 2018-09-10:
@mikeller Thanks, is working now with BF 3.5
Regards.
[image]
[image]

closed 2018-09-10 by FasaArg