betaflight/betaflight #6505: BF 3.4 wrong negative altitude telemetry value

kev018, 2018-08-01:
**Describe the bug**
Since betaflight 3.4 and also with 3.5RC1 I have a big wrong negative altitude telemetry value.

With BF 3.3.3 and before its all normal. 

It does not matter if the barometer is on or off, as well armed or disamed.
The configurator displays the right altitude at the GPS or sensor tab, also the OSD value is correct.

[image]

**To Reproduce**
The negative altitude value is displayed as soon as GPS fix is active.

**Flight controller configuration**
```
# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.4.0 Jul  8 2018 / 04:19:05 (cce8a59be) MSP API: 1.39

board_name MATEKF405
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature AIRMODE

# beeper
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -SYSTEM_INIT
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 5 1400 2100 0
aux 1 1 1 900 1300 0
aux 2 2 1 1300 1700 0
aux 3 27 4 1700 2100 0
aux 4 46 2 1700 2100 0
aux 5 13 0 1600 2100 0
aux 6 35 4 1600 2100 0
aux 7 39 3 1400 2100 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1100
rxfail 7 s 1100
rxfail 8 s 1100

# master
set acc_calibration = 33,-55,-410
set align_mag = CW270FLIP
set mag_calibration = 124,-78,59
set rssi_channel = 12
set rc_interp_ch = RPYT
set rc_smoothing_type = FILTER
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = KILL
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 80
set osd_cap_alarm = 1200
set osd_alt_alarm = 0
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2113
set osd_throttle_pos = 2433
set osd_current_pos = 2504
set osd_mah_drawn_pos = 2512
set osd_gps_speed_pos = 2329
set osd_gps_lon_pos = 2530
set osd_gps_lat_pos = 2544
set osd_gps_sats_pos = 2104
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2094
set osd_altitude_pos = 2294
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2497
set osd_disarmed_pos = 2410
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set debug_mode = FFT
set vtx_channel = 5
set vtx_freq = 5820
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0
set dterm_setpoint_weight = 120
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 32
set i_pitch = 65
set d_pitch = 23
set p_roll = 30
set i_roll = 60
set d_roll = 21
set i_yaw = 60
set d_yaw = 20

# rateprofile
rateprofile 0

set roll_rc_rate = 121
set pitch_rc_rate = 121
set yaw_rc_rate = 115
set roll_srate = 78
set pitch_srate = 78
set yaw_srate = 78
set tpa_rate = 15
set tpa_breakpoint = 1050
```

**Setup / Versions**
FC: Matek F405-CTR
GPS: BN880
Crossfire Micro TX, Nano Rx
Taranis x9D Plus (2.2.1stable)

mikeller, 2018-08-02:
Looks like #5764 broke this.

@kev018: Can you please try this:

[betaflight_3.4.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2252563/betaflight_3.4.0_MATEKF405.hex.zip)

mikeller, 2018-08-02:
(#6205 will fix this.)

kev018, 2018-08-02:
Many thanks, it works!! With which version will the fix be officially available?
Is the version from you safe to use? So only the altitude fix changed from the stable 3.4 version?

mikeller, 2018-08-03:
I'll try to get this into 3.5.

PR #6513 (2018-08-06): Fixed altitude display with CRSF telemetry.
Fixes #6505.
closed 2018-08-06 by mikeller