betaflight/betaflight #6507: BF 3.3.3 Target DLF4 BEEPER RX_LOST not working

ukbren, 2018-08-02:
Setting up beeper on my DAL F4 with BF 3.3.3
Beeper seems to work normally in all other respects but not on RX_LOST (when I turn off the Tx).  LED on FC flashes but no beeps are emitted.  Beeper is configured to beep on a Tx switch, 

```# diff

# version
# Betaflight / DALRCF405 (DLF4) 3.3.3 May 31 2018 / 13:00:10 (3565182d3) MSP API: 1.38

# name
name UKBren

# resources

# mixer

# servo

# servo mix

# feature
feature OSD
feature ANTI_GRAVITY

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# map

# serial
serial 0 64 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0
aux 1 2 0 1300 1700 0
aux 2 13 1 1800 2100 0
aux 3 28 0 1800 2100 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set rssi_channel = 16
set blackbox_device = NONE
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 32
set vbat_warning_cell_voltage = 33
set small_angle = 45
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_warnings = 8
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim1 = -11775
set osd_vbat_pos = 2513
set osd_rssi_pos = 2504
set osd_tim_1_pos = 2102
set osd_flymode_pos = 2082
set osd_mah_drawn_pos = 450
set osd_craft_name_pos = 34
set osd_warnings_pos = 329
set osd_disarmed_pos = 2283
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_max_curr = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 2
set vtx_freq = 5806

# profile
profile 0

set p_pitch = 55
set p_roll = 37

# rateprofile
rateprofile 3

set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 76

```

etracer65, 2018-08-02:
Are you testing with the USB cable connected?  You have `beeper -ON_USB` set so you won't hear the beeps when connected to USB.

Also `RX_LOST` is only enabled after you've armed at least once since there's no point for the lost model beeper to go off unless you've actually flown.  Try arming and disarming then try turning the radio off.  **AWAYS TEST WITHOUT PROPS!**

ukbren, 2018-08-03:
Ugh! ;)
You know, I never realised it only worked once the quad had been armed - after all these years!  It's working fine.  How embarrassing... 🗡 
- bren

closed 2018-08-03 by ukbren