betaflight/betaflight #6512: Why is there no update for the CL_Racing F4 flight controller? I bought it but there has been no updates for the firmware.

Ekoran, 2018-08-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

DieHertz, 2018-08-03:
Clracingf4

closed 2018-08-03 by DieHertz

mikeller, 2018-08-04:
@Ekoran: This target has been renamed, it's now called 'CLRACINGF4' in the drop down in configurator.