betaflight/betaflight #6518: Speedybee DAC pin not working for camera control

TwoToneEddy, 2018-08-05:
Hi,
I'm looking into why when I connect camera control to the DAC pad of this flight controller the camera control doesn't work. Looking at the speedybeef4 target I see no mention of a dac pin. Is this where I should expect to find it? Will it be called DAC or LED? 

Looking at another board which also has a DAC pin the matekf405 I see it has 2 LED pins defined, is one of these mapped to the DAC pad? 

If it simply hasn't been defined in the target i'm happy to define it and test it, i just need to make sure i'm defining it correctly as I see no examples on other boards.
Thanks

mikeller, 2018-08-05:
@TwoToneEddy: Camera control via a DAC pin is not yet supported. You need to use PWM on a pin that has a timer (i.e. a motor output or LED_STRIP pin on the SPEEDYBEEF4).

TwoToneEddy, 2018-08-05:
Perfect, thanks for the clarification. I'll use the led.

closed 2018-08-05 by TwoToneEddy