betaflight/betaflight #6520: FAILSAFE GPS RESCUE ... ¿DISARM AT HOME POSITION?

luismiguelsi, 2018-08-05:
If the drone returns to the home position and still does not receive a signal from the receiver, it should be disarmed.

aowi7280, 2018-08-15:
in RC2, I just had mine disarm about 30' up. Bent 2 motors and cracked a GoPro lens. I think it should go to the selected failsafe procedure. Either drop or land.

SchonRichtig, 2018-08-15:
Activated by Switch, or as Failsave?
Try: get failsafe_procedure if its not GPS-RESCUE (drop or land)

type: set failsafe_procedure = GPS-RESCUE

if you set it to land within the configutator i think 
it will also land in the CLI but this is not the GPS Rescue funktion!

aowi7280, 2018-08-15:
I had rescue on a switch to test it. Then I had failsafe set that aux channel to "on" when it activates. I will set it up for failsafe like you said. But I hesitate to test it since I just rebuilt the quad.

SchonRichtig, 2018-08-15:
Ah, ok, 

so it was active for about a second, than the stage 2 failsave was aktivated.
Havn`t tried the new configurator yet, maybe it ist possible to set it to GPS-Rescue 
using the new version.
Otherwise use the cli.
And make no changes using the configurator at the failsave tab, this will set it back to drop or land.
I check it in cli to go for sure.

aowi7280, 2018-08-15:
No. The failsafe should not have been involved. I was just using the switch to test gps rescue.

callagga, 2018-08-22:
When RESCUE_SANITY_ON is set AND you are within 100m, what does betaflight do in the case:
(a) you trigger rescue mode with a taranis switch
(b) failsafe occurs

For example for (a) does nothing happen, but for (b) the quad falls out of the sky?

Ralfde, 2018-08-29:
I used a switch for the RESCUE mode and had no failsafe. The copter fall down, too.
But I was maybe within 100 m distance. I don't want to make such tests the first time with a higher distance.

callagga, 2018-08-31:
I think RESCUE_SANITY_ON is on by default in 3.5 which would explain it. You could turn it off whilst testing.

closed 2018-10-08 by stale[bot]