betaflight/betaflight #6524: Tags without "v" prefix missing from GitHub dropdown menu

Cleric-K, 2018-08-06:
Ever since the "v" in front of the tag names was dropped (from 3.4 onwards), the tags do not appear in the drop down list on the front page:
[image]

I did some tests and the problem stems from the fact that the drop down is truncated to around 100 tags. The ordering scheme puts the names starting with digits _after_ all these starting with "v" and thus they are being truncated from the list.

This makes it a little cumbersome to jump to the code of the latest tags - they can be reached from the releases section only.

As a fun fact I was even late to upgrade to 3.4 because I periodically checked the tags drop down and looked for 3.4 but it never appeared. Only when more and more videos appeared in youtube about it I got suspicious and checked the releases section and found out that it was already out weeks ago :D

Was the drop of the "v" intentional?

mikeller, 2018-08-07:
Yes, the drop was intentional, since it does not serve any purpose (it was a requirement for release tags in SemVer 1.0.0, but has since been dropped), and only introduces ambiguity as to what should have a `v`, and what not (e.g. should a release branch have the `v` even though the spec does not say so).

The way that GitHub does populate the tag drop down with the first 100 tags *alphabetically* is a usability problem really, it should rather be the *latest* 100 tags. An issue has been raised with GitHub.

Cleric-K, 2018-08-07:
Ok,
thank you!

closed 2018-08-07 by Cleric-K