betaflight/betaflight #6530: Motors can spin up/twitch in CLI when pasting a diff

brycedjohnson, 2018-08-07:
I've had this happen quite a few times, and I've seen some reports of it.

For the Facebook people: https://m.facebook.com/groups/253033415182436?view=permalink&id=446527169166392&ref=m_notif&notif_t=group_comment

I think it is because the default fresh flash is oneshot and switching to dshot in the CLI we'll start sending dshot commands that get misinterpreted as oneshot signals.  If the motor power is on you'll get twitches and maybe spinups.  

Props off, motor power off is the best solution, but people will keep in running into it anyways...

Traveling right now so I haven't verified it on 3.5.  but I had it happen in 3.4 with a Nero board.  (I needed lipo power because the 5v on the USB is pulled down to much by vtx, etc to program it)

flint723, 2018-08-08:
I've seen it in 3.5 too.  On Spracingf3evo (this is the only board I have that requires lipo power for usb to work).

jflyper, 2018-08-08:
There are limited set of configuration items that take effect immediately, but we don't know which causes the issue.

We need a set of two configs (diff output okay) that triggers it.

flint723, 2018-08-08:
Copying in this diff onto a full chip erase flash of 3.5 does it for me.  
[diff 7-27-2018.txt](https://github.com/betaflight/betaflight/files/2268867/diff.7-27-2018.txt)

mikeller, 2018-08-08:
Admittedly this is not nice. But on the other hand I think this is very hard to completely eliminate - powering your ESCs / motor drivers while an incorrect motor output signal is selected has the potential to result in unwanted motor behaviour in a lot of different and hard to control situations.

Therefore, I think the only 'fix' for this is to add even more warnings that users should not under any circumstances power up their ESCs with props on while their config is incomplete.

jflyper, 2018-08-08:
Another option would be to add a "motor output enable" variable which initializes to off after full chip erase flash or `defaults`.

I think iNav does this.

brycedjohnson, 2018-09-07:
Still an issue stalebot.

closed 2018-10-14 by stale[bot]